MarlinFirmware/Marlin #15579: Variable Gcode + M851

xero83, 2019-10-16:
Hello,
I'm looking to do the following:
G28
G1 Z0
Manually I put the probe to the measure of the bed and put to take the present value with M114 of the value of Z, and this to declare it to a varialbe #POSITION.
Then make an M851 Z#POSITION + M500.
I do not know if it is possible to do it or there is a simpler way to perform this task.

Thank you very much.

boelle, 2019-10-16:
is this a question or a bug?

InsanityAutomation, 2019-10-16:
its a FR to have M851 store the current position as new offset and apply showing a new POS of 0

xero83, 2019-10-17:
that's what I'm looking for, which is not how to take the position variable to apply it to M851 Z....

But I don't know how to do this

closed 2020-06-30 by github-actions[bot]