MarlinFirmware/Marlin #15580: [HELP] Screen Problems

uok8252, 2019-10-16:
Hello everyone,
I got a problem about marlin 1.19 and 2.0
My screen doesn't work correctly.
I upload some videos about my problem.
( I install Repetier and screen works correctly but i want use marlin so please help me about problem)
[https://youtu.be/b7X2DoF9ORY](url)

DiogoLambert, 2019-10-17:
I am having the same problem after updating the arduino!

boelle, 2019-10-17:
is this a bug or question?

uok8252, 2019-10-17:
> is this a bug or question?

We don't know.

AnHardt, 2019-10-17:
The usual timing problems. Use the delays.

boelle, 2019-10-17:
@AnHardt so this is an config error and not a marlin failure as such?

boelle, 2019-10-17:
@uok8252  there are also these places you can get help

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

uok8252, 2019-10-17:
> The usual timing problems. Use the delays.

i can't fix can you help me please.
Here is my codes.
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3741353/Marlin-bugfix-2.0.x.zip)

boelle, 2019-10-17:
@uok8252 did you try any of the places i suggested?

uok8252, 2019-10-18:
> @uok8252 did you try any of the places i suggested?

Yes i try discord but no body can help me ? My english not so good , so i understand little bit

uok8252, 2019-10-18:
I upload with vscode and its fixed

boelle, 2019-10-18:
oki, will close then :-D

closed 2019-10-18 by boelle