MarlinFirmware/Marlin #15581: Homing at the edge of the Bed instead of the middle[BUG] (short description)

tifrei, 2019-10-16:
Hi,

I compiled MArlin 2.0 a few times now but always run into the issue that when i run G28 it homes all the way to the right of the ender 3.  

Bed_size is set correctly.
Probe is working, but when running g29 it tries to run outside the bed.
I hope anybody can help me cause this is driving me crazy

Config.H is attached.

[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/3735641/Archiv.zip)

GhostlyCrowd, 2019-10-16:
Have you verified you have safe homing enabled?

Kokorone, 2019-10-16:
Personally I dont directly define Z_SAFE_HOMING, which you have in your config as 115. That should be fine, but I'd go with this to make sure it always homes in the center.

`#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif`

ManuelMcLure, 2019-10-16:
This could be caused by incorrect microsteps set on your TMC2208 drivers. It looks like you've got it configured with 20 tooth pulleys on 1.8 degree motors with 16 microsteps - if instead your drivers are running at 8 microsteps it will home X and Y fine (although faster than expected) and then move twice the distance.
What does ``M122`` report for the TMC2208 settings, specially msteps?

thisiskeithb, 2019-10-17:
Did you remove all the jumpers under the X & Y drivers?

tifrei, 2019-10-17:
Hi @ all,

Yes the jumper are removed. I had this setup working and just changed to the newest firmware cause i changed the hotend mount and needed to edit the probe offset anyways. @ManuelMcLure  ill check that whenever i get home. But its weired that Y goes in to the mid point and only X goes all the way to the right.

thisiskeithb, 2019-10-17:
Did you reset your `EEPROM` by sending `M502` and `M500` after updating your firmware?

tifrei, 2019-10-17:
I did not i thought it will be deleted by flashign the new firmware? 
Im gone try all of your tipps later and let you know! Thanks for the response !

DkOuvE, 2019-10-17:
May be related to bug #15577.

closed 2019-10-17 by tifrei