MarlinFirmware/Marlin #15587: [BUG] USB COMPOSITE for BTT E3 SKR DIP not working

Bansh1990, 2019-10-17:
### Description
I am flash last commit, use in platformio.ini STM32F103R_bigtree_USB, but arter flashing cartrider over usb not working on my board

what I do wrong?

reloxx13, 2019-10-17:
the env is bugged, remove the non-usb block and rename the usb enviroment to the non usb name.

i did this here, you can copy this ini file over yours.
https://github.com/reloxx13/Marlin/blob/my-marlin/platformio.ini

Duplicate:
https://github.com/MarlinFirmware/Marlin/issues/15306

reloxx13, 2019-10-17:
> EEPROM works fine after that?

EEPROM in flash did not work for me. sd works.

i disabled `FLASH_EEPROM_EMULATION` in the `Marlin\src\pins\stm32\pins_BIGTREE_SKR_MINI_E3.h` file to get eeprom (on sd card) working.

boelle, 2019-10-17:
so this is maybe not a marlin bug?

reloxx13, 2019-10-17:
> 
> 
> so this is maybe not a marlin bug?

what else if the plaformio.ini is wrong?

boelle, 2019-10-17:
upps... in that case yes, a PR sent to bugfix 2.0.x would be the fastest fix it

brianredbeard, 2019-10-17:
This seems to be related to the issue @Bansh1990 was experiencing in https://github.com/MarlinFirmware/Marlin/issues/15269

Bansh1990, 2019-10-18:
umm... I'm tried, 

> the env is bugged, remove the non-usb block and rename the usb enviroment to the non usb name.
> 
> i did this here, you can copy this ini file over yours.
> https://github.com/reloxx13/Marlin/blob/my-skr-mini-e3/platformio.ini
> 
> Duplicate:
> [BUG 2.0.x]Autobuild STM32F103R_bigtree_USB not detected #15306

it doesn't help me( i'm try to waiting a fix.

boelle, 2019-10-26:
@Bansh1990 many updates have been done, does it still not work? and is https://github.com/MarlinFirmware/Marlin/issues/15269 a dublicate of this one?

boelle, 2019-11-03:
Duplicate of #15269

closed 2019-11-03 by boelle