MarlinFirmware/Marlin #15589: [BUG] LPC1768: automatic detection of upload drive does not work anymore

Bergerac56, 2019-10-17:
The "Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py" does not seem to work for me anymore for me.

I get:
Unable to find destination disk (a bytes-like object is required, not 'str')
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Could it be related to the version of Python installed on my PC? (python 2.7.13 and 3.7.4)

tvixen, 2019-10-17:
yeps... won't work with Python 3.7. Delete and re-install Python 2.x

boelle, 2019-10-17:
so we can close this one allready?

ManuelMcLure, 2019-10-17:
Given that Python 2 is fast approaching end-of-life, I would posit that this needs to be fixed. However, this may be a PlatformIO issue rather than a Marlin one.

tvixen, 2019-10-17:
> Given that Python 2 is fast approaching end-of-life, I would posit that this needs to be fixed. However, this may be a PlatformIO issue rather than a Marlin one.

It is... nothing to do with Marlin software!

ManuelMcLure, 2019-10-17:
Actually, it is a Marlin bug - the code that tries to find the upload port is a python script in Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py - this is our script and needs to be fixed for Python 3 compatibility.

tvixen, 2019-10-17:
Well, we can call it a dependency if u like. And then again, everything is :o)

blueeagle69, 2019-10-17:
I'm glad this is a known issue.
Grabbed the latest build, and had the same error when compiling for my SKR 1.3.
I just checked my system, and according to that I have 2.7 and 3.7 installed.
Using VS Code.

p3p, 2019-10-18:
Are all of you with the error on Windows?

Powerprobot, 2019-10-18:
Yes i have the same problem... I make a dirty fix, so i comment out the "drive found" routine and add an hardcoded drive letter. Please fix this automatic search. (The PlatformIO has the same issue, but it's not your workplace...)

Bergerac56, 2019-10-18:
I did just the same to fix temporarily this issue. And... yes it is on windows using vs-code and platformio. But I tried also atom with the same result. 
BTW, I removed python 3.7 and cheched that I had only 2.7 installed. It still does not work

tvixen, 2019-10-18:
You have to remove them both. And then install 2.7 again.

boelle, 2019-10-18:
will close this one as it has nothing to do with marlin

blueeagle69, 2019-10-18:
Fixed! Thank you 😀

ManuelMcLure, 2019-10-19:
This does have to do with Marlin. Our script does not handle Python 3, Python 2 is near end-of-life. The script needs to be fixed to work with Python 3.

p3p, 2019-10-19:
Twas fixes, check the PR above.

Bergerac56, 2019-10-20:
I agree, boelle is... a bit over enthusiastic ;). Thanks for the fix.

Bergerac56, 2019-10-20:
Mmmm I just checked the new bugfix with the "python" fix. It does not work for me. I get:

Unable to find destination disk ('utf-8' codec can't decode byte 0xff in position 10: invalid start byte)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Did I forget something? (clean up,...?)

boelle, 2019-10-20:
did you do an update all? not sure if that will fix it

Bergerac56, 2019-10-20:
Yes I did. But I will retry with an entirely fresh release so I will be sure it is not on my side ;)

p3p, 2019-10-20:
No you're running the patch, you can tell because it fails to decode as utf-8 which was the fix .. I have no idea what the correct character encoding is if its not utf-8, I tested on Windows 10 with python 2.7, 3.5 and 3.8, (after installing it -.- I don't use the OS).

Does Windows just use one character encoding for running programs in a subprocess or is it language dependant .. this is a rabbit hole I don't really want to go down ><

Patag, 2019-10-20:
Same here. I had to "hard code" the drive letter in platformio.ini like this : 
```
[env:LPC1768]
.
.
.
upload_port       = G:
.
.
`
```
Just change your drive letter

Bergerac56, 2019-10-20:
@p3p Hello Chris. Not sure I understand what you mean. I used a pc where a fresh installation of python 3.8 was just done. I installed VScode and platformio (this PC was never used for 3D stufs), took the last bugfix (with your fix. To be sure, I did not start from my gitkracken test environment as I have a lot of personnal changes there), imported just my config and pins files and compiled. I got the message indicated above.
This does not block me anyway as I force the drive letter in platformio file. But the automatic function was pretty cool ;)

Patag, 2019-10-20:
And forcing drive letter is just a workaround, not a fix !

blueeagle69, 2019-10-21:
@Bergerac56 Weird.
I simply edited the lines of code shown in the commit with my current build and that worked fine with the drive letter.

Powerprobot, 2019-10-21:
So, i test it... I am not a python programmer. I start with the Commodore64....
But... It works! I found the solution for Windows with Python 3.74
Here is the complete upload script. I don't know to make a commit.

`#
# sets output_port
#  if target_filename is found then that drive is used
#  else if target_drive is found then that drive is used
#
from __future__ import print_function

target_filename = "FIRMWARE.CUR"
target_drive = "REARM"

import os
import platform
current_OS = platform.system()
Import("env")

def print_error(e):
    print('\nUnable to find destination disk (' + e + ')\n' \
          'Please select it in platformio.ini using the upload_port keyword ' \
          '(https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) ' \
          'or copy the firmware (.pio/build/' + env.get('PIOENV') + '/firmware.bin) manually to the appropriate disk\n')

try:
    if current_OS == 'Windows':
        #
        # platformio.ini will accept this for a Windows upload port designation: 'upload_port = L:'
        #   Windows - doesn't care about the disk's name, only cares about the drive letter
        #

        #
        # get all drives on this computer
        #
        import subprocess
        # typical result (string): 'Drives: C:\ D:\ E:\ F:\ G:\ H:\ I:\ J:\ K:\ L:\ M:\ Y:\ Z:\'
        driveStr = str(subprocess.check_output("fsutil fsinfo drives"))
        # typical result (string): 'C:\ D:\ E:\ F:\ G:\ H:\ I:\ J:\ K:\ L:\ M:\ Y:\ Z:\'
        # driveStr = driveStr.strip().lstrip('Drives: ') <- Doesn't work in other Languages as English. In German is "Drives:" = "Laufwerke:"
        FirstFound = driveStr.find(':',0,-1)         # Find the first ":" and
        driveStr = driveStr[FirstFound + 1 :-1]      # truncate to the rest
        # typical result (array of stings): ['C:\\', 'D:\\', 'E:\\', 'F:\\',
        # 'G:\\', 'H:\\', 'I:\\', 'J:\\', 'K:\\', 'L:\\', 'M:\\', 'Y:\\', 'Z:\\']
        drives = driveStr.split()
        
        upload_disk = 'Disk not found'
        target_file_found = False
        target_drive_found = False
        for drive in drives:
            final_drive_name = drive.strip().rstrip('\\')   # typical result (string): 'C:'
            try:
                volume_info = str(subprocess.check_output('cmd /C dir ' + final_drive_name, stderr=subprocess.STDOUT))
            except Exception as e:
                continue
            else:
                if target_drive in volume_info and target_file_found == False:  # set upload if not found target file yet
                    target_drive_found = True
                    upload_disk = final_drive_name
                if target_filename in volume_info:
                    if target_file_found == False:
                        upload_disk = final_drive_name
                    target_file_found = True

        #
        # set upload_port to drive if found
        #

        if target_file_found == True or target_drive_found == True:
            env.Replace(
[... 55 lines cut ...]
            try:
                filenames = os.listdir('/Volumes/' + drive + '/')   # will get an error if the drive is protected
            except:
                continue
            else:
                if target_filename in filenames:
                    if target_file_found == False:
                        upload_disk = '/Volumes/' + drive + '/'
                    target_file_found = True
        #
        # set upload_port to drive if found
        #

        if target_file_found == True or target_drive_found == True:
            env.Replace(
                UPLOAD_PORT=upload_disk
            )
            print('\nupload disk: ', upload_disk, '\n')
        else:
            print_error('Autodetect Error')

except Exception as e:
    print_error(str(e))
`
[upload_extra_script.zip](https://github.com/MarlinFirmware/Marlin/files/3752146/upload_extra_script.zip)

Bergerac56, 2019-10-22:
@Powerprobot Well done. It works independently of the windows OS language. (Checked with french, english and flemish)

closed 2019-10-22 by Bergerac56

Powerprobot, 2019-10-22:
@boelle : can you commit my changed code? I don't know how to commit the code. Thanks!

boelle, 2019-10-22:
yes, just tell me precise what file i should replace and i can give it a shot

Powerprobot, 2019-10-22:
`..Marlin\Marlin\src\HAL\HAL_LPC1768\upload_extra_script.py`

I added the file here as ZIP
Thanks!
[BIGTREE_TFT35_V3.0_TVL1.zip](https://github.com/MarlinFirmware/Marlin/files/3757533/BIGTREE_TFT35_V3.0_TVL1.zip)

boelle, 2019-10-22:
https://github.com/MarlinFirmware/Marlin/pull/15637 once merged should take care of this one