MarlinFirmware/Marlin #15591: Marlin / Anet - Skew Compensation not working

mtlred, 2019-10-17:
Hello,

I've spent a lot of time on this site and others trying to correct this. 

I printed the test print as described in the Marlin firmware (see pic), I then inputted all the measurements (after doubling them) into the firmware.  

I un-commented what I believe is supposed to be: 

#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 319
  #define XY_DIAG_BD 318
  #define XY_SIDE_AD 199

I also made sure that  #define XY_SKEW_FACTOR 0.0 was commented out. (i did try with the skew factor with the other lines commented out and that did not work either). 

I also tried with the following both commented and un-commented. 

#define SKEW_CORRECTION_GCODE

When I upload the firmware, with the #define SKEW_CORRECTION_GCODE and #if ENABLED(SKEW_CORRECTION) I can go to the terminal and type M852 and it will return a value (in my case 0.00322, which i double checked with the formula on wolfram).

It seems like octopi/the board is seeing the skew correction but it does not change my prints. 

I've also tried hard coding it in the print file, also no results.  I even incresed the value by 300% to see if there was a difference and there was not, however typing M852 in the terminal would bring up the increased value. 

Any help would be greatly appreciated. 

[image]

boelle, 2019-10-17:
have you tried any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

mtlred, 2019-10-17:
I have.  Should I post there?

boelle, 2019-10-17:
yep, the issue list here is for bugs and feature requests

boelle, 2019-10-18:
i assume that you found help and will close this one

closed 2019-10-18 by boelle