MarlinFirmware/Marlin #15592: [BUG] FlagBits X and Y locations swapped

Jimbo145, 2019-10-17:
inline void unmark(const xy_int8_t &xy) and inline bool marked(const xy_int8_t &xy)  have their x and y locations transposed.

`template <int W, int H>
  struct FlagBits {
    typename IF<(W>8), uint16_t, uint8_t>::type bits[H];
    void fill()                                   { memset(bits, 0xFF, sizeof(bits)); }
    void reset()                                  { memset(bits, 0x00, sizeof(bits)); }
    void unmark(const uint8_t x, const uint8_t y) { CBI(bits[y], x); }
    void mark(const uint8_t x, const uint8_t y)   { SBI(bits[y], x); }
    bool marked(const uint8_t x, const uint8_t y) { return TEST(bits[y], x); }
    inline void unmark(const xy_int8_t &xy)       { unmark(xy.y, xy.x); }
    inline void mark(const xy_int8_t &xy)         { mark(xy.y, xy.x); }
    inline bool marked(const xy_int8_t &xy)       { return marked(xy.y, xy.x); }
  };`

boelle, 2019-10-18:
this one could be fixed faster if you send a PR to 2.0.x

boelle, 2019-10-22:
@Jimbo145 is this a fix for something?

Roxy-3D, 2019-10-23:
@Jimbo145 

Jim, did you notice this one might be flipped also?
```
    inline void mark(const xy_int8_t &xy)         { mark(xy.y, xy.x); }
`````
I changed all three...   Can you confirm they are all correct?

Thank You!

Jimbo145, 2019-10-23:
@Roxy-3D

Can confirm, all three needed the fix.

Thanks for hammering it out!

Jimbo145, 2019-10-23:
Fixed with 3642a12

closed 2019-10-23 by Jimbo145