MarlinFirmware/Marlin #15596: Compilation error for LPC1768 after platformio update

Dracoit, 2019-10-18:
During compilation I am getting the following error:

Can not remove temporary directory `c:\3d printer\Marlin-bugfix-2.0.x-SKR V1.1\.pioenvs`. Please remove it manually to avoid build issues
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; board: nxp_lpc1768)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.1.3
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.4.6
  File "C:\3d printer\Marlin-bugfix-2.0.x-SKR V1.1\Marlin\src\HAL\HAL_LPC1768\upload_extra_script.py", line 16

    print '\nUnable to find destination disk (' + e + ')\n' \

                                              ^

SyntaxError: invalid syntax

================================================================= [FAILED] Took 2.98 seconds =================================================================

Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        IGNORED
megaatmega1280        IGNORED
at90usb1286_cdc       IGNORED
at90usb1286_dfu       IGNORED
DUE                   IGNORED
DUE_USB               IGNORED
DUE_debug             IGNORED
LPC1768               FAILED    00:00:02.975
LPC1769               IGNORED
melzi                 IGNORED
melzi_optiboot        IGNORED
rambo                 IGNORED
sanguino_atmega644p   IGNORED
sanguino_atmega1284p  IGNORED
STM32F1               IGNORED
STM32F4               IGNORED
ARMED                 IGNORED
mks_robin             IGNORED
teensy35              IGNORED
malyanm200            IGNORED
esp32                 IGNORED
fysetc_f6_13          IGNORED
============================================================ 1 failed, 0 succeeded in 00:00:02.975 ============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Please help me solve this issue I am using SKR v1.1 board.
thanks

AnHardt, 2019-10-18:
> Can not remove temporary directory c:\3d printer\Marlin-bugfix-2.0.x-SKR V1.1\.pioenvs. Please remove it manually to avoid build issues

Please do exactly that. Likely you have this directory open in a explorer or shell.

GhostlyCrowd, 2019-10-18:
Building fine here.  As stated above you should do as its asking in the error output.

boelle, 2019-10-18:
will close this as its clearly an user error and not a issue with marlin

and a "fix" has been provided

closed 2019-10-18 by boelle

Dracoit, 2019-10-19:
I did as instructed in the error, but it shows the same error. 
Also, the folder (.pioenvs)  is created every time I try to build it.
I am using the following version of marlin 2.0: https://github.com/Msq001/BIGTREETECH-SKR-V1.1/tree/master/firmware/Marlin-bugfix-2.0.x-SKR%20V1.1.

Thanks

AnHardt, 2019-10-19:
You are not using the firmware from here.
We don't have exact kollagen about what their are doing.
We don't have any influence upon whats happening there.

Go to where you got it - and ask for help there.

minoger, 2019-10-19:
i have opened an issue here: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/102