MarlinFirmware/Marlin #15600: [BUG] BTT SKR Mini E3 V1.2 doesn't boot when flashed

m4rl0ne, 2019-10-19:
I flashed plain Marlin on my SKR mini E3 board. Now, it does not boot up. Screen backlight is on, nothing shows up. PC won't recognize it either. When I flash the default Firmware from BTT (which is also some version of Marlin 2.0) the board works fine. Since I want to customize some things in the fw I cant use that version.

Steps to reproduce:

1. Download latest Marlin 2.0 branch
2. Configure like they tell you in their BTT guide for the E3 mini
3. Flash and nothing works

Expected behavior:
Board boots up.

Actual behavior:
Screen is lit, board is unresponsive.

Configuration is standard for Ender 3.
TMC_DEBUG is disabled, EEPROM_SETTINGS disabled, using CR10_STOCKDISPLAY as LCD.

I'm suspecting 2 things: either the TMC2209 library has a bug or it has something to do with the lcd.

boelle, 2019-10-19:
did you read the issue template?
where are your configs?
where do you get marlin from?

shitcreek, 2019-10-19:
How are you flashing the board? LPC and STM32 platforms are flashed via SD card, not USB.

AmiSMB, 2019-10-21:
Have use tried compiling without USB_COMPOSITE enabled?

thedark20, 2019-10-24:
Hi all
I got the same issues following configuration from BTT. I use BLTouch 3.1 on servo pins + endstop on Zendstop. If i try another config, i got the board boot up but TMC connection error message on the lcd. 
Here my config 
[Marlin-2.0.x-SKR-Mini-E3-V1.2 config files.zip](https://github.com/MarlinFirmware/Marlin/files/3766864/Marlin-2.0.x-SKR-Mini-E3-V1.2.config.files.zip)
[image]

boelle, 2019-10-24:
i dont think stuff from BTT (or any other fork) is supported

only stuff from marlin

thedark20, 2019-10-24:
I use last marlin 2.0 branch. but it is possible that is a problem from last revision of their BTT board.

boelle, 2019-10-24:
i only wrote that because there was a few cases where people reported an issue and they used a fork of marlin and not direct :-D

but yeah hardware issue is most often the issue, or misunderstandings in how things should be done

thedark20, 2019-10-24:
hi
found this ; https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/tree/bltouch-improvements

that's work fine.

boelle, 2019-10-24:
that is a fork and we cant support that

shitcreek, 2019-10-24:
@thedark20 that fork is a bit behind Marlin's. I can't see what all has been changed. BLTouch and pins files have no changes and the configuration appears pretty standard. Also using Z min endstop plug for the BLTouch is the standard for most boards.
Do you know which changes resolved your issue so that we may get the fix merged.

m4rl0ne, 2019-10-25:
Thanks to everyone for replying. I fixed my issue by simply updating the platformio plugin for atom. Now it works flawlessly.

closed 2019-10-25 by m4rl0ne

thinkyhead, 2019-10-25:
Note that Atom will no longer be supported by PlatformIO going forward.