MarlinFirmware/Marlin #15601: [BUG] (short description)

manarsak, 2019-10-19:
Hi, I used to show the remaining time to print with the percentage, but the progress bar is not moving and remains constant and the percentage is 99 What is the problem
[image]

boelle, 2019-10-20:
have you tried asking these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

manarsak, 2019-10-20:
My friend this is a problem in marlin and not a question

boelle, 2019-10-20:
you did forget to attach config files as the issue template told you to so i assumed it was a question

boelle, 2019-10-24:
could you attach your config files as zip?

boelle, 2019-10-30:
closing this one due missing config files

we can reopen later

closed 2019-10-30 by boelle