MarlinFirmware/Marlin #15603: Heaters turning off while still printing (USB/Octoprint)

randellhodges, 2019-10-19:
### Description

The same gcode file, printed from SD card works as expected, but printed from Octoprint causes the heaters to turn off early*.

It acks like, when printing from Octoprint it buffers commands. While processing the incoming commands, it is like the temperature commands are processed immediately while there is still stuff in the buffer/planner(?). This is causing my heaters to turn off while there is still a few layers left to print.

### Steps to Reproduce

1. Print gcode from SD, verify the fans do not turn off until all movement is completed.
2. Upload the gcode to Octoprint (use safe mode so we know plugins aren't the problem)
3. Print and watch the temperature graph (or listen from the extruder to start skipping). You'll see that the heaters turn off while the print head is still moving and trying to extrude.

NOTE: My BLOCK_BUFFER_SIZE is set to 64. I think if this was lower, the problem would get "hidden" because the planner/buffer would be smaller? (Just a guess)

**Expected behavior:** 

**Actual behavior:** The heaters to stay on all the moves are executed

#### Additional Information

My configuration can be found here:
https://github.com/randellhodges/Marlin/tree/bugfix-2.0.x/Marlin

The configuration files and example gcode is also attached.
[Supporting-Files.zip](https://github.com/MarlinFirmware/Marlin/files/3747095/Supporting-Files.zip)
Note: I have attached the Octoprint serial log. I turned it on after the print had started and once it turned off and my extruder started skipping, I hard powered the printer off (that is why you see the serial errors at the end)

randellhodges, 2019-10-19:
I just noticed there is a M400. I am going to try that in my end g-code and see if that makes a difference.

http://marlinfw.org/docs/gcode/M400.html

boelle, 2019-10-20:
@randellhodges so not a bug  after all?

ManuelMcLure, 2019-10-20:
It's arguable whether this is a bug or not. Being a non-blocking temperature change command it's obvious that whatever comes after the temperature change command should happen immediately, what's less obvious is whether the command should begin changing the temperature before the move buffer is clear. In any case, there's a pretty simple workaround.

boelle, 2019-10-20:
i would say without knowing how to verify, that if marlin does as the code says then its more like how the user is expecting things to work vs. how its made to work

p3p, 2019-10-20:
I would argue that non blocking infers that commands that come after are executed immediately rather than waiting for completion, but it still shouldn't break gcode command sequence.

There are very few immediate commands that do break sequence, emergency stop being the only one I can think of at the moment.

randellhodges, 2019-10-20:
In a strict sense, not a bug. Like I mentioned, I would imagine if I used a smaller buffer, I wouldn't have "noticed" it.

To expand on what @ManuelMcLure said, I was not expecting this non-blocking temperature change to happen out of order. I expected everything before it to have been executed before it. Of course, being non-blocking, once executed it shouldn't wait for the temperature change before continuing.

Print over USB with the large buffer. You'll see the temperature change happens well enough in advance that the filament cools and the extruder starts skipping because it cannot push the filament.

I feel the M400 is a workaround. I don't think people should need to douse their sliced gcode to ensure a temperature change does't "start" before expected. That is what I consider the bug.

reloxx13, 2019-10-25:
On the other hand, eg.: if you print and notice your temp is too low and you want to set it higher, you want it immediately and not after the movement buffer is done.
Guess this was the intention why its changing temp as soon as it arrives.

IMHO its better have it immediately as waiting to end movements.

randellhodges, 2019-10-25:
That is a good point and probably why it was implemented as such.

For a temperature adjustment, that is probably the best implementation. Turning off the heaters, however, should probably be dependent on the movement buffer.

A couple ways that could be done, but all have pros and cons

1) A new code for turning a heater off (seems redundant and slicers will probably never use it)
2) A parameter to the existing codes to turn a heater off (same opinion as the above)
3) Look at the temperature on the command and if it is a 0 (off), then buffer the temp change so it gets executed after the movements that have been buffered before it, otherwise implement the change.

The 3rd option sounds like the idea situation. Slicers wouldn't need to change. Not sure in what situations this would be detrimental. Any edge case scenarios that this idea would case havoc?

I cannot think of a situation where you would want to turn the heats off (0) while there are still movements for printing in the buffer. 

Hmmm, actually, if we really really wanted to implement this and cover "all the bases" we'd probably need it to be a threshold. So if you had multiple tools and want to keep a nozzle warm (that is under the extrusion temperture) then buffer that. Or maybe if prevent cold extrusion is turned on, any temp change below that is buffered. I guess this would be more of a conversation to have if we decide to go down this path.

boelle, 2019-11-03:
@randellhodges i assume this is still an issue in 2.0.x ?

randellhodges, 2019-11-04:
@boelle It is of my opinion that it is still an issue. However, at this point, whether it is an issue or a feature request remains to be seen.

It points down to, yes, it shouldn't block, but it it shouldn't jump the queue either, especially turning heaters off. (I don't think adjusting fan speeds jump the queue).

Some post processing in Cura and an M400 command solves my immediate needs but I think further discussion or action is needed. I think @p3p hit it on the head with https://github.com/MarlinFirmware/Marlin/issues/15603#issuecomment-544279885 but I'm not sure if that means the behavior is should change or not. I would try to submit a PR but this is pretty deep in the code and my understanding/skills are not up to par.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

SBachman, 2019-12-09:
I'm also experiencing this after increasing my buffer. I don't have enough time to marge the latest update to test so for now I'll add an M400 to the start of my end Gcode.

randellhodges, 2019-12-09:
@boelle No change that I am aware of. I'd have to pull my M400 out and try my calibration cube. 

I'm in a bit disorganized at the moment so it would be a while before I can try. I haven't noticed any code changes around the planner that would change the behavior, so while I cannot say 100% for certain the error still exists, I am about 90% sure ;)

boelle, 2020-01-10:
@randellhodges no pressure and i assume that it will be a while before you can retest with bugfix 2.0.x

boelle, 2020-01-28:
if you have a small buffer it wont be very noticable

the default size would barely make it noticeable, but the larger the buffer the more you notice

i did a consult with @Vertabreak and he have a 5 sec gab  at end of print but that is due to the buffer size set higher than default

so will close this one, we can always reopen if we are wrong

closed 2020-01-28 by boelle