MarlinFirmware/Marlin #15605: upload_extra_script for the LPC1768 HAL broken (str vs. byte output)

sbaeder, 2019-10-19:
minor issue in the Marlin-bugfix-2.0.x

To work correctly with latest Python (in VSCode) you need to convert the output from bytes to str.  Not sure how this will work on non-english systems, since the stripping of the word "Drives:" may not be 100% kosher.

```
34c34
<         driveStr = subprocess.check_output("fsutil fsinfo drives")
---
>         driveStr = str(subprocess.check_output("fsutil fsinfo drives"), 'utf-8')
47c47
<                 volume_info = subprocess.check_output('cmd /C dir ' + final_drive_name, stderr=subprocess.STDOUT)
---
>                 volume_info = str(subprocess.check_output('cmd /C dir ' + final_drive_name, stderr=subprocess.STDOUT), 'utf-8')

```

Certainly not critical, and easy to force with env variable, but would be nice to fix...

Thanks - Keep up the good work!!

gloomyandy, 2019-10-20:
I'm pretty sure that there has already been a fix for this...
https://github.com/MarlinFirmware/Marlin/pull/15597

But there may still be some issues!

sbaeder, 2019-10-20:
Thanks - I dod some searching before opening this, but didn't find it based on script name or platform...That fix looks fine - I just did the conversion a bit differently, but it should fix it.  I'll pull newest code to double check, but for now, we can close this out.

closed 2019-10-20 by sbaeder