MarlinFirmware/Marlin #15607: use MIGHTYBOARD_REVE with ff Creator II,E0 can't heating

yajiedesign, 2019-10-20:
i am use Marlin to  flashfogre Creator II (MIGHTYBOARD bord).
the E0 can't heating.Besides E0 fan ,E1,stepper,estop,lcd,sdcard all work fine.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3747330/config.zip)

boelle, 2019-10-24:
you still have this issue?

have you checked for bad wires etc?

yajiedesign, 2019-10-26:
still,i am back to use Sailfish.

boelle, 2019-11-03:
do you test with marlin from time to time or have you given up?

very hard to solve problems as we most likely dont have the hardware

etet100, 2019-11-09:
Hi. I have exactly the same problem. High state on mosfet gate is too low (about 0,5V instead of 5V). It looks like it's not configured as output. I have just replaced Marlin.cpp -> setup() with simple 

  #define EX1_HEAT_PIN        6   // H3
  #define EX1_FAN_PIN         7   // H4
  #define EX2_HEAT_PIN       11   // B5
  #define EX2_FAN_PIN        12   // B6

  pinMode(EX1_FAN_PIN, OUTPUT);
  pinMode(EX1_HEAT_PIN, OUTPUT);
  pinMode(EX2_FAN_PIN, OUTPUT);
  pinMode(EX2_HEAT_PIN, OUTPUT);

  while (1) { 
    digitalWrite(EX1_FAN_PIN, HIGH);
    digitalWrite(EX1_HEAT_PIN, HIGH);
    digitalWrite(EX2_FAN_PIN, HIGH);
    digitalWrite(EX2_HEAT_PIN, HIGH);
    _delay_ms(1000);
    digitalWrite(EX1_FAN_PIN, LOW);
    digitalWrite(EX1_HEAT_PIN, LOW);
    digitalWrite(EX2_FAN_PIN, HIGH);
    digitalWrite(EX2_HEAT_PIN, HIGH);
    _delay_ms(1000);
  }

and i see all my hardware is working just fine. 

[MightyboardCreatorConfig.zip](https://github.com/MarlinFirmware/Marlin/files/3827624/MightyboardCreatorConfig.zip)

etet100, 2019-11-09:
I found commit which introduced this issue: 

293b890940a884480715be9a83d6ba30941d2bf7
Apply soft limits to G2/G3 final target (#15156)
4 Sep 2019

`apply_motion_limits(raw)` in G2_G3.cpp

boelle, 2019-11-11:
@shitcreek @thinkyhead maybe this is an easy fix ?

@etet100 do you know how to fix it? if so maybe submit a PR to 2.0.x

etet100, 2019-11-12:
I have no idea how to fix it. I don't even know how this function can affects heaters. It's weird.

boelle, 2019-11-21:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

etet100, 2019-11-21:
And this one https://github.com/MarlinFirmware/Marlin/issues/15054

boelle, 2019-11-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-24 by boelle

etet100, 2019-11-24:
Lack of activity? I hope you mean lack of activity from developers side.

I spent many days trying to figure why my printer is completely bricked. I think the problem is here: 

https://github.com/MarlinFirmware/Marlin/pull/14125/commits
https://github.com/MarlinFirmware/Marlin/pull/14125/commits/b1988683cf11a0510393e798787104196a98bcc2

Original author of FFCP integration (@mikeshub) removed info about missing pins declarations from pins_MIGHTYBOARD_REVE.h. It's fine until we use external dependencies like SlowSoftI2CMaster. This library still uses incomplete arduino declarations with only 70 pins. Trying to set any pin above 69 to input is overflowing digital_pin_to_port_PGM and cousing such unexpected behaviour.

boelle, 2019-11-24:
> I hope you mean lack of activity from developers side.

nope i did mean lack of activity, do you expect people that work for free to solve issue quick?
what did you pay when you started to use marlin?

etet100, 2019-11-24:
I don't expect anything from anyone. I just assumed that authors care about quality of this project.

boelle, 2019-11-24:
they do, but they work for free

and they do not have every bit of hardware so they cant test everything themselfes

etet100, 2019-11-24:
I'm aware of this. I have done a lot to help solve this issue. And that's why it hurts a bit that issue has been closed without any answer from developers.

boelle, 2019-11-24:
if i remember correct you fixed the issue right?

etet100, 2019-11-26:
It's fine. Thanks.

chepo92, 2020-01-15:
Is this solved/pr'ed? I'm having this issue

etet100, 2020-01-15:
You have to do this

```
/* in [arduino dir]/hardware/arduino/avr/variants/mega/pins_arduino.h	  /**
change:	   * Software serial
#define NUM_DIGITAL_PINS            70	   */
to:	
#define NUM_DIGITAL_PINS            80	  #define X_SERIAL_TX_PIN    16
  #define X_SERIAL_RX_PIN    17
to digital_pin_to_port_PGM add at the end:	
const uint8_t PROGMEM digital_pin_to_port_PGM[] = {	  #define Y_SERIAL_TX_PIN    18
....	  #define Y_SERIAL_RX_PIN    19
        PG      , // PG 4 ** 70 **	
        PG      , // PG 3 ** 71 **	  #define Z_SERIAL_TX_PIN    41
        PJ      , // PJ 2 ** 72 **	  #define Z_SERIAL_RX_PIN    66
        PJ      , // PJ 3 ** 73 **	
        PJ      , // PJ 7 ** 74 **	  #define E0_SERIAL_TX_PIN   40
        PJ      , // PJ 4 ** 75 **	  #define E0_SERIAL_RX_PIN   67
        PJ      , // PJ 5 ** 76 **	
        PJ      , // PJ 6 ** 77 **		#define E1_SERIAL_TX_PIN   37
        PE      , // PE 2 ** 78 **	  #define E1_SERIAL_RX_PIN   68
        PE      , // PE 6 ** 79 **	
};	#endif
to digital_pin_to_bit_mask_PGM  add at the end:	
const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[] = {	
....	
        _BV( 4 )        , // PG 4 ** 70 **	
        _BV( 3 )        , // PG 3 ** 71 **	
        _BV( 2 )        , // PJ 2 ** 72 **	
        _BV( 3 )        , // PJ 3 ** 73 **	
        _BV( 7 )        , // PJ 7 ** 74 **	
        _BV( 4 )        , // PJ 4 ** 75 **	
        _BV( 5 )        , // PJ 5 ** 76 **	
        _BV( 6 )        , // PJ 6 ** 77 **	
        _BV( 2 )        , // PE 2 ** 78 **	
        _BV( 6 )        , // PE 6 ** 79 **	
};	
to digital_pin_to_timer_PGM add at the end:	
const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {	
....	
        NOT_ON_TIMER    , // PG 4 ** 70 **	
        NOT_ON_TIMER    , // PG 3 ** 71 **	
        NOT_ON_TIMER    , // PJ 2 ** 72 **	
        NOT_ON_TIMER    , // PJ 3 ** 73 **	
        NOT_ON_TIMER    , // PJ 7 ** 74 **	
        NOT_ON_TIMER    , // PJ 4 ** 75 **	
        NOT_ON_TIMER    , // PJ 5 ** 76 **	
        NOT_ON_TIMER    , // PJ 6 ** 77 **	
        NOT_ON_TIMER    , // PE 2 ** 78 **	
        NOT_ON_TIMER    , // PE 6 ** 79 **	
};	
*/
```

chepo92, 2020-01-15:
Thanks @etet100 ! will try it, do you know if that change would break the compatibility with atmega2560? (as I see you are changing #define X_SERIAL_TX_PIN  16 #define X_SERIAL_RX_PIN    17 )

Also, I noticed Mightyboard has a #define CUTOFF_RESET_PIN   16   // H1 that controls the optocouples that feed the hotend mosfets, if that pin is not high (managed to enable with M42 P16 S255) the hot ends won't work, any ideas to control it with firmware each time you set temperature ? or maybe leave it always enabled? another way would be adding it as start gcode as a temporary solution

etet100, 2020-01-15:
Mistake when copy - pasting. It should be

```
/* in [arduino dir]/hardware/arduino/avr/variants/mega/pins_arduino.h
change:
#define NUM_DIGITAL_PINS            70
to:
#define NUM_DIGITAL_PINS            80
to digital_pin_to_port_PGM add at the end:
const uint8_t PROGMEM digital_pin_to_port_PGM[] = {
....
        PG      , // PG 4 ** 70 **
        PG      , // PG 3 ** 71 **
        PJ      , // PJ 2 ** 72 **
        PJ      , // PJ 3 ** 73 **
        PJ      , // PJ 7 ** 74 **
        PJ      , // PJ 4 ** 75 **
        PJ      , // PJ 5 ** 76 **
        PJ      , // PJ 6 ** 77 **
        PE      , // PE 2 ** 78 **
        PE      , // PE 6 ** 79 **
};
to digital_pin_to_bit_mask_PGM  add at the end:
const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[] = {
....
        _BV( 4 )        , // PG 4 ** 70 **
        _BV( 3 )        , // PG 3 ** 71 **
        _BV( 2 )        , // PJ 2 ** 72 **
        _BV( 3 )        , // PJ 3 ** 73 **
        _BV( 7 )        , // PJ 7 ** 74 **
        _BV( 4 )        , // PJ 4 ** 75 **
        _BV( 5 )        , // PJ 5 ** 76 **
        _BV( 6 )        , // PJ 6 ** 77 **
        _BV( 2 )        , // PE 2 ** 78 **
        _BV( 6 )        , // PE 6 ** 79 **
};
to digital_pin_to_timer_PGM add at the end:
const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {
....
        NOT_ON_TIMER    , // PG 4 ** 70 **
        NOT_ON_TIMER    , // PG 3 ** 71 **
        NOT_ON_TIMER    , // PJ 2 ** 72 **
        NOT_ON_TIMER    , // PJ 3 ** 73 **
        NOT_ON_TIMER    , // PJ 7 ** 74 **
        NOT_ON_TIMER    , // PJ 4 ** 75 **
        NOT_ON_TIMER    , // PJ 5 ** 76 **
        NOT_ON_TIMER    , // PJ 6 ** 77 **
        NOT_ON_TIMER    , // PE 2 ** 78 **
        NOT_ON_TIMER    , // PE 6 ** 79 **
};
*/
```
It's not likely to break break compatibility.

I'm not sure what is CUTOFF_RESET_PIN. Sorry.

chepo92, 2020-01-15:
It works! Thanks!

michaelVisimid, 2020-01-15:
@etet100 

"Lack of activity? I hope you mean lack of activity from developers side."

That was me and I did that out of the kindness of my heart. You're welcome. 

Coming soon I am going to get off my ass and design a PCB to replace my hand soldered adapter so you can use any modern 3d printing board with the stock display. My Creator Pro (Now creator EXTREME) runs on an SKR 1.3