MarlinFirmware/Marlin #15608: [BUG] G29 Not making use of HYBRID_THRESHOLD?

sn4k3, 2019-10-20:
### Description

This problem is happening to me only in stealthchop mode, meaning spreadCycle works fine.
The motor X & Y is stalling once it starts first point at 10000mm/m. Bed probing continues but of course on wrong coordinates, and it seems to stall only once per G29. After lower to 8000mm/m sometimes goes good others don’t, stalling at last row sometimes. Both speeds pass my set HYBRID_THRESHOLD which make me think G29 not making use of it? While printing infills, I run it at 150mm/s to 200mm/s max without any problem (no stall nor skips).
To fix the problem I set XY_PROBE_SPEED to 7000mm/m, which pass a little bit my limit but works ok so far without stalling. 
While slower speeds have no problem, probing with bltouch is slow and I like to give it a boost.
As a feature request marlin could have a define to temporary set spreadCycle while probing, something like #define TMC_FORCE_SPREADCYCLE_G29, which put XYZ  on spreadCycle and restore previous value after probing

```
#define STEALTHCHOP_XY
#define X_HYBRID_THRESHOLD     100  // [mm/s]
#define Y_HYBRID_THRESHOLD     100  // [mm/s]
#define BLTOUCH_HS_MODE
#define XY_PROBE_SPEED 10000
```

BastR, 2019-10-20:
If I remember correctly, the G29 uses hybrid threshold, you may be right it's a bug. In the meantime you can try to force spreadcycle mode in your startgocde before G29, like:

M569 S0 X Y Z ;Disable stealthChop
G29 ;Autolevel
M569 S1 X Y Z ;Enable stealthChop

sn4k3, 2019-10-20:
> If I remember correctly, the G29 uses hybrid threshold, you may be right it's a bug. In the meantime you can try to force spreadcycle mode in your startgocde before G29, like:
> 
> M569 S0 X Y Z ;Disable stealthChop
> G29 ;Autolevel
> M569 S1 X Y Z ;Enable stealthChop

I know i can use that script to overcome the problem, but that would violate defaults, for example if my profiles are shared and people use spreadCycle instead on their machine - That would put them in stealthChop

shitcreek, 2019-10-23:
Mind following the template and provide your config files so we would have more information to work with. What sort of acceleration are you working with? What TMC drivers do you have?

Just as a side note, TMC2209 has SpreadCycle as **a part** of StealthChop2 - so `TMC_FORCE_SPREADCYCLE_G29` may be in conflict with how it functions.

sn4k3, 2019-10-23:
Sure, both configs attached.
Is with TMC2209
#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 100, 5000 }

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3763890/Configuration_adv.zip)

> Mind following the template and provide your config files so we would have more information to work with. What sort of acceleration are you working with? What TMC drivers do you have?
>
> Just as a [...]

thisiskeithb, 2019-10-24:
> #define `STEALTHCHOP_XY`
> #define `HYBRID_THRESHOLD`
> #define `X_HYBRID_THRESHOLD`     100  // [mm/s]
> #define `Y_HYBRID_THRESHOLD`     100  // [mm/s]
> #define `BLTOUCH_HS_MODE`
> #define `XY_PROBE_SPEED` 10000

We run similar Ender-3/SKR 1.3/BLTouch configs, except I am using Watterott 5160s.

I just rebased [my config](https://github.com/thisiskeithb/Marlin/tree/df8dc4afb1582227365bfeec5c04079ef9b50398) and the drivers switch between StealthChop and SpreadCycle modes without issue.

I made sure that VSCode and PIO (+ libraries) were updated as well before building and ran `M502`/`M500` after updating the firmware.

boelle, 2019-10-24:
@thisiskeithb so you have no issues?

thisiskeithb, 2019-10-24:
> so you have no issues?

Correct.

boelle, 2019-10-24:
so most likely a misunderstanding on how to config?

boelle, 2019-10-30:
will close this one as it mostly just a misunderstanding on how config should be

we can always reopen

closed 2019-10-30 by boelle

thisiskeithb, 2019-10-30:
I was using commits current as of yesterday and everything worked as expected.