MarlinFirmware/Marlin #15609: [BUG] G92 software endstop offsets persist after G28 on delta

Knifa, 2019-10-20:
### Description

Noticed this as Cura uses G92 to set a Z offset (positive, in my case). Successive prints without rebooting my delta printer caused the hot end to move up and up with each print, ruining the print. 
 It seems G92 changes the software endstops that are required for deltas. I can't seem to find any G-code that would reset the software endstops.

My printer homes to Z_MAX and is running on bugfix-2.0.x @ 9470554f6343ac971636cc19cc1cdc59e9bb1a9a.

### Steps to Reproduce

Bit of an example below. Basically, I would expect G28 to reset these endstops (i.e., hardware endstops trump software endstops), or there to be some other way to reset the software endstops.
```
G28
M211
; Soft endstops: ON  Min:  X-116.00 Y-116.00 Z0.00   Max:  X116.00 Y116.00 Z289.04

G92 Z-100
M211
; Soft endstops: ON  Min:  X-116.00 Y-116.00 Z-315.23   Max:  X116.00 Y116.00 Z-26.19

G28
M211
; Soft endstops: ON  Min:  X-116.00 Y-116.00 Z-315.23   Max:  X116.00 Y116.00 Z-26.19
```

What Cura does (simplified):
```
G28
G0 Z0.38 ; (i.e., 0.2mm layer height + 0.18mm)
G92 Z0.2
G0 X... Y... Z0.2 E... ; etc
; ....
G0 X... Y... Z0.4 E... ; etc --- actually Z=0.58mm in reality
```
**but successive calls to G92 stack since G28 doesn't reset them so if I run another print with this**:
```
G28 ; doesn't reset the original G92
G0 Z0.38 ; this is actually Z=0.56
G92 Z0.2 ; printer thinks its at Z=0.2 but it's Z=0.56
G0 X... Y... Z0.2 E... ; etc, making a huge mess on the bed
```

#### Additional Information

Not sure if I'm maybe just missing some magic G-code here but I am pretty sure this worked before. Surely I don't need to reboot my printer every time to reset these endstops?

Config: [config.zip](https://github.com/MarlinFirmware/Marlin/files/3748107/config.zip)

Knifa, 2019-10-20:
Some rough notes:

G92 changes `position_shift` https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/geometry/G92.cpp#L92
```               
position_shift[i] += d;       // Other axes simply offset the coordinate space
```

`position_shift` is defined in `motion.cpp` --- the comments seem to indicate it should reset on G28 but G92 is literally the only thing that changes it. https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/motion.cpp#L180

``` 
// The distance that XYZ has been offset by G92. Reset by G28.
xyz_pos_t position_shift{0};
```

Knifa, 2019-10-20:
Can confirm that `NO_WORKSPACE_OFFSETS` resolves this issue.

boelle, 2019-10-22:
will close this one as it seems "fixed"

we can always reopen if there is a confirmed bug

closed 2019-10-22 by boelle

Knifa, 2019-10-22:
@boelle: Cheers! Do you think it would be reasonable to make a PR to disable workspace offsets on deltas by default? IMO this is pretty surprising behaviour.

boelle, 2019-10-22:
yes, Scott will let you know if something in a PR is not correct

remember to link this issue in the PR description