MarlinFirmware/Marlin #15610: [BUG] Apply rotation function always gives Z = 1

Evgeny-SPB, 2019-10-20:
### Description

After "Add custom types for position" https://github.com/MarlinFirmware/Marlin/commit/50e4545255605eb506c20eb107270038b0fe7bdb commit planar leveling is broken.

```
// Apply a rotation to the matrix
void vector_3::apply_rotation(const matrix_3x3 &matrix) {
  const float _x = x, _y = y, _z = z;
  *this = matrix.vectors[0] * _x + matrix.vectors[1] * _y + matrix.vectors[2] * _z;
}
```
If we change new function
```
*this = matrix.vectors[0] * _x + matrix.vectors[1] * _y + matrix.vectors[2] * _z;
```
 to old style
```
x = _x * matrix.vectors[0][0] + _y * matrix.vectors[1][0] + _z * matrix.vectors[2][0];
y = _x * matrix.vectors[0][1] + _y * matrix.vectors[1][1] + _z * matrix.vectors[2][1];
z = _x * matrix.vectors[0][2] + _y * matrix.vectors[1][2] + _z * matrix.vectors[2][2];
```
everything seems to work as expected.

### Steps to Reproduce

1. #define AUTO_BED_LEVELING_3POINT in Configuration.h
2. #define M114_DETAIL in Configuration_adv.h
2. G28
3. G29
4. M114 D
5. Z will be always equal to 1.0.

#### Additional Information

I think that this problem arises because of multiplying array on a float is not working as expected.

boelle, 2019-10-22:
please do, Scott will let you know if its not ok

remember to add link for this issue in PR description

shitcreek, 2019-10-24:
I've created the PR for it. I kept`*this` just incase it is needed elsewhere. Tested to work with 3Points and ABL.

Evgeny-SPB, 2019-10-24:
@shitcreek thank you! But I already created one :) https://github.com/MarlinFirmware/Marlin/pull/15636

@p3p also already created review request from @thinkyhead for this issue

I think both styles are good and the both will help marlin become better 💪
I don’t know what style of the function better conform to Marlin coding standards :)
I like my style better, because it is more human readable IMHO 😂

Evgeny-SPB, 2019-10-24:
I don’t know, why I can’t see information about my pull my request in this issue. Usually if I reference an issue information appears here automatically 🤔

shitcreek, 2019-10-24:
@Evgeny-SPB didn't see your PR before. Good work. Either style works for me, mine just makes sure *this is set just in case it is referred to at some other junction.

Evgeny-SPB, 2019-10-24:
Issue is solved with PR https://github.com/MarlinFirmware/Marlin/pull/15636, which is already merged. I close this issue.

closed 2019-10-24 by Evgeny-SPB