MarlinFirmware/Marlin #15614: [FR] MMU2S support.

TB1405, 2019-10-21:
At the moment we have a working feature for MMU2. i use it with a SKR Pro and it works very nice. 
The new MMU2s unit uses a IR Sensor in the extuder to detected when the filament loaded to the btech gears. 
In MMU2 version you have to calibrate the length of your bowden tube. 

it works but is a little bit tricky to calibrate and if there is a load or unload error the printer doesnt detect this. 

so it would be very helpful to integrate the latest mmu2.h and .cpp from prusa into the new marlin version. 
I've tried this myself but unfortunately I didn't make it because my knowledge is very limited.

tedukes, 2019-10-26:
I'm wondering if you could make a 'Y' adapter cable for the runout sensor with on set of wires going the MMU board and the other to the SKR board?

I have the SKR 1.3 and m really interested in the MMU if it can work properly.

brianmoody, 2020-01-19:
Excellent somone already had this same thought!  Basically looking to introduce the IR sensor capability at the extruder to Marlin.

I also would appreciate this functionality if someone has the time and skill to implement it!

Many thanks in advance!

notching, 2020-01-20:
Lets get this done please!

Vertabreak, 2020-01-20:
this is open source software the best way to get something done is to roll up your sleeves code it and submit a pull request. a dev would need one to perform testing on to even attempt anything, i don't even know anyone that owns a MMU of any kind.

tedukes, 2020-01-20:
I bought a used one off eBay, works like a champ!! Not real good at making cables with those small pin connectors. I'd try but I need to look into getting an extra cable for the runout sensor on the MMU2S, just in case I fubar things.

closed 2020-06-30 by github-actions[bot]