MarlinFirmware/Marlin #15615: Use marlin in EFF but ALSO have a Heated Bed

safaenet, 2019-10-21:
Hi, i use marlin on ramps 1.4.
my question is: Is there a way to use marlin in EFF mode but still have heated bed via an external MOSFET ?
My goal is to use both D8 and D9 pins for Fans (D10 for Heater) and using an external MOSFET connect a heated bed to it (Don't wanna use RRD fan extender)
If So what codes in marlin must be edited to do that ?

boelle, 2019-10-21:
have you asked any of these places first?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-10-21:
oki, i will close this one  because the list is for bugs only

if this turns out to be a bug we can reopen it or you can create a new

closed 2019-10-21 by boelle

AnHardt, 2019-10-21:
Due to the diversification into EBF, EFF, ... and use in multiple different pins files as an include RAMPS pin file is a bit complicated to read. For the predefined configurations it's working fine (mostly).
When is comes to a special custom pinning, forgetting about the pre configurations is much simpler then  altering them.
Comment out the definitions based on the pre configurations and just define the 3-4 needed pins unconditional by yourself.