MarlinFirmware/Marlin #15616: [BUG] SPI_ENDSTOPS Beta Feature enables Sensorless Homing on All three Axis regardless of config

adamnock, 2019-10-21:
### Description

The SPI_ENDSTOPS beta feature has been coded so that Sensorless Homing is enabled across all of X, Y and Z axis' regardless of user configuration.
This is evident in 

Conditionals_post.h
`#if ENABLED(SPI_ENDSTOPS)
    #define X_SPI_SENSORLESS X_SENSORLESS
    #define Y_SPI_SENSORLESS Y_SENSORLESS
    #define Z_SPI_SENSORLESS Z_SENSORLESS
  #endif`

and
endstops.h
`#if ENABLED(SPI_ENDSTOPS)
      typedef struct {
        union {
          bool any;
          struct { bool x:1, y:1, z:1; };
        };
      } tmc_spi_homing_t;
      static tmc_spi_homing_t tmc_spi_homing;`

### Steps to Reproduce

0. Checkout latest bugfix 2.0 branch
1. Enable SPI_ENDSTOPS where the physical printer has on of X, Y or Z as a physical endstop.
2. Compile and upload to board
3. For safety, using M119 observe that the physically switched endstop will not trigger regardless of input state and inversion config.

**Expected behavior:** 
As per old DIAG pin sensorless homing, if a homing sensitivity is not enabled for an Axis, that axis wont have TMC Sensorless enabled

**Actual behavior:** 
Endstops expect SPI fed states and are no longer triggerable even if no sensitivity is defined

#### Additional Information

I had a fair go at solving this, but im not C++ savvy and changing the endstops.h code to set struct bools based on #_STALL_SENSITIVITY is beyond me. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3752065/Marlin.zip)

teemuatlut, 2019-10-22:
```cpp
  #define X_SENSORLESS  (AXIS_HAS_STALLGUARD(X)  && defined(X_STALL_SENSITIVITY))
  #if ENABLED(SPI_ENDSTOPS)
    #define X_SPI_SENSORLESS X_SENSORLESS
  #endif
```
`X_SPI_SENSORLESS` expands into `(AXIS_HAS_STALLGUARD(X)  && defined(X_STALL_SENSITIVITY))`
but is not defined unless `SPI_ENDSTOPS` is also defined.

The struct does have all the axes are in the struct but the components are used within the compile conditions.

boelle, 2019-10-24:
so is this a bug or misconfig or misunderstanding?

closed 2019-10-24 by adamnock

boelle, 2019-10-24:
well we blame the maker of the solder joint :-D no need to say names

good that you found it