MarlinFirmware/Marlin #15619: [FR] cylindrical coordinate moveset

Schild0r, 2019-10-21:
Hey there, I don't know how the movement for a delta is calculated so this might be easy or not so much.
I would like to have a cylindrical coordinate moveset for my delta 
a) when moving the tool via LCD
b) maybe even for gcode control (?)

Since deltas have a cylindrical build space it is more convenient IMO to have cylindrical coordinates i.e. R (radius) P (Phi/Angle) Z (height) instead of X, Y and Z. Also this would make sure that the toolhead can be moved to the very edge of the complete build plate without overshooting the Delta_Printable_Radius or "stair stepping" to the edge (at every angle).

AnHardt, 2019-10-22:
The build-volume of a DELTA printer is **NOT** a cylinder. (https://docs.google.com/viewer?a=v&pid=forums&srcid=MTgyNjQwODAyMDkxNzQxMTUwNzIBMDc2NTg4NjQ0MjUxMTE1ODY5OTkBakVPUkdfbDNkVEVKATAuMQEBdjI&authuser=0)
It's just the firmware what clips the shape (in the lower part) to a cylinder to simplify the calculation.

A polar coordinate system has no runtime advantage for a delta printer.
A slightly simplified calculation when clipping (x²+y²<R² vs r<R) is more than compensated by the  sin()/cos() transformations  to calculate xy from Phi.

![delta clipping outated](
https://cloud.githubusercontent.com/assets/4294448/10342278/4f71a69a-6ce6-11e5-87a5-9996ff8c0269.png)
![Delta buildvolume](
https://cloud.githubusercontent.com/assets/211931/13879216/94970e44-ed16-11e5-87d9-9b6bbff7efa8.png)

Schild0r, 2019-10-22:
> The build-volume of a DELTA printer is **NOT** a cylinder.

Ok maybe I was not being precise. The reachable space of course is not a cylinder But since most deltas use a round baseplate the build space most of the time is considered a cylinder as one can also see in your pictures. 

> A polar coordinate system has no runtime advantage for a delta printer

I see, but I don't request that movement (during normal runtime) is calculated in cylindrical (or polar + z ) coordinates. I just request a feature that makes it possible to move the printhead in plane in polar coordinates instead of x, y. Doesn't matter whether this is actually calculated with polar coordinates in mind or by some transformation or arbitrary method.

closed 2020-06-30 by github-actions[bot]