MarlinFirmware/Marlin #15620: [2.0.X] MKS robin 2 compile error with "SS_PIN not defined!" and some other errors

userosos, 2019-10-21:
Cant make the firmware for mks robin 2 use Atom and Platformio.
I can not make full log because i can not copypaste from console platformio ide log.
[errors.txt](https://github.com/MarlinFirmware/Marlin/files/3752528/errors.txt)

userosos, 2019-10-21:
Full log!
[log.txt](https://github.com/MarlinFirmware/Marlin/files/3752657/log.txt)

boelle, 2019-10-24:
do you know what is wrong and how to fix? if so we could fix it faster

my quick guess is that the root of the problem is SS_PIN not defined

boelle, 2019-10-30:
@userosos any news?

boelle, 2019-11-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-03 by boelle