MarlinFirmware/Marlin #15622: [build Issue] Still unable to build for Malyan m200 / STM32

Eno423, 2019-10-21:
### Description

Tried all listed methods including x000005's standalone Arduino IDE package & still no dice

### Steps to Reproduce

1) follow any guide to build for STM32
have it fail while compiling

**Expected behavior:** [What you expect to happen]
that it completes
**Actual behavior:** [What actually happens]
with Platformio all kinds of errors w/ the STM32xxxx.h
with Arduino IDE portable package from x000005 ,  board 3dprinter is unknown in IDE 1.90
#### Additional Information
Sorry to re-hash this thread or a similar thread, but I really want to get this built so that i can use this board; else I may go to another board/screen but rather not

xC0000005, 2019-10-21:
You don't have to use my portable install anymore.
[The instructions here ](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Malyan/M200) are good with one caveat - use 1.6.1 for now. Theres a timer change that makes it incompatible with newer releases and I'm in the middle of moving houses and don't have time to update the HAL for that.

The bug with the board disappearing has bitten a few people, I'm guessing it has to do with portable specifically.
Install whatever build you want, add ST's board manager, 1.6.1, and both V1 (STM32F103) and V2 (STM32F070) should be available.

thinkyhead, 2019-10-21:
If 1.6.1 works for you, we will update the README to point out the requirement.

boelle, 2019-10-24:
so this one is kind of solved?

Eno423, 2019-10-25:
well not yet. 

So i ran through a few builds, and not sure if its a PEBKAC issue or the step 15 is incorrect.

[quote]
When compilation is done you've built the firmware. The next stage is to flash it to the board. To do this look for a line like this: "path/to/bin/arm-none-eabi-objcopy" -O binary "/path/to/Marlin.ino.elf" "/path/to/Marlin.ino.bin" The file Marlin.ino.bin is your firmware binary. M200 (v1-3) and M300 printers require flashing via SD card. Use the SD card that came with the printer if possible. The bootloader is very picky about SD cards. Copy Marlin.ino.bin to your SD card under three names: firmware.bin, update.bin, and fcupdate.flg.
[/quote]

but I get the warning about the CMSIS, but figure that isnt anything to worry about. 
It tells me how much program space, then dynamic memory, & compile says "done"

However I do not have a marlin.ino.elf ; I do have a Marlin.ini.MalyanM200_F103cb.hex
Would this be the file I should be renaming 3x?

xC0000005, 2019-10-25:
If you have chosen to export a binary it should created the .bin.

Eno423, 2019-10-26:
This is resolved. My issue with just the HEX. was because I had the "programmer" set in Arduino to AVR still. I switched it to Arduino ISP &  we good now : )

Eno423, 2019-10-28:
well, almost.

Looks like I have to adjust something. my Z works, but all others just buzz.

xC0000005, 2019-10-28:
Turn off stepper disable and see if it works.

Eno423, 2019-10-29:
> Turn off stepper disable and see if it works.
> […](#)

Are you referring to the following in the config_adv.h?
Current settingsVVVV
// Time can be set by M18 and M84.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // Set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E true

should I change them all to (w/ exception of Z)
// Time can be set by M18 and M84.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X false
#define DISABLE_INACTIVE_Y false
#define DISABLE_INACTIVE_Z true  // Set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E false

xC0000005, 2019-10-29:
Yes. If the steppers move on USB gcode but buzz on the menu try turning that off.

boelle, 2019-11-03:
@Eno423 many updates have passed in the last 6 days, is this still an issue in 2.0.x ?

Eno423, 2019-11-03:
I am going to check tonight. I pulled a recent copy down (last Friday) and tried to compile but now the Arduino ide is locking up right near the end.

I did try to turn off the setting in the config adv but didn't seem to work.

boelle, 2019-11-03:
have you tried vscode+platform.io?

Eno423, 2019-11-03:
Yeah apparently the only way to build for this malyan is Arduino ...platformio doesn't work 

I would love to use platformio on atom ornvsxode vs Arduino

xC0000005, 2019-11-03:
PIO will work if the new hardware timer PR is merged.

boelle, 2019-11-03:
> platformio on atom ornvsxode

why atom?

Eno423, 2019-11-03:
fixed the typo. on atom or vscode.

I only mention atom as I use that on my ESP82xx outlets

boelle, 2019-11-03:
:-) i thought atom was long dead

Eno423, 2019-11-04:
so now trying to use vscode+platformio

Same/similar error as atom+platformio

I get a bunch of Board/Board.h no such file
<details><summary>Click to expand build errors</summary>
`LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM>
|-- <Wire> 1.0
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\bkp_f1.c.o
In file included from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\HardwareSerial.h:39:0,
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\dma.c.o
                 from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\HardwareSerial.cpp:33:
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\dma_f1.c.o
C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\boards.h:39:10: fatal error: board/board.h: No such file or directory
 #include <board/board.h>
Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\exc.S.o
          ^~~~~~~~~~~~~~~Compiling .pio\build\STM32F103CB_malyan\FrameworkArduino\libmaple\exti.c.o

compilation terminated.
C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\HardwareTimer.cpp:32:10: fatal error: board/board.h: No such file or directory
 #include <board/board.h>           // for CYCLES_PER_MICROSECOND
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103CB_malyan\FrameworkArduino\HardwareTimer.cpp.o] Error 1
*** [.pio\build\STM32F103CB_malyan\FrameworkArduino\HardwareSerial.cpp.o] Error 1
In file included from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\IPAddress.cpp:20:
C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:39:10: fatal error: board/board.h: No such file or directory
 #include <board/board.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\wirish.h:54:0,
                 from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\Arduino.h:30,
                 from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\Stream.cpp:23:
C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:39:10: fatal error: board/board.h: No such file or directory
 #include <board/board.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\ext_interrupts.cpp:38:0:
C:\Users\1996P\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\boards.h:39:10: fatal error: board/board.h: No such file or directory
 #include <board/board.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103CB_malyan\FrameworkArduino\Stream.cpp.o] Error 1
*** [.pio\build\STM32F103CB_malyan\FrameworkArduino\IPAddress.cpp.o] Error 1
*** [.pio\build\STM32F103CB_malyan\FrameworkArduino\ext_interrupts.cpp.o] Error 1`

Eno423, 2019-11-04:
I am wondering if its due to the difference between Marlins platformio file & what platformio has listed?

Marlins file has the board named "STM32F103CB_malyan"

and platformio lists the name as "malyanm200_f103cb " & also suggest to name it that via their site
https://docs.platformio.org/en/latest/boards/ststm32/malyanm200_f103cb.html?utm_source=platformio&utm_medium=piohome

boelle, 2019-11-04:
i would make a wild guess here and say that a board is selected that is not there

Eno423, 2019-11-05:
Unfortunatlely 

I am not sure where "there" is

So i went hog wild and copied a board directory from the variants directory & started dropping it everywhere.

re-ran the build.

new error. Variant.h : no such file..

so same thing., copy/pasted everywhere I could think of, 

now another build util/atomic.h : no such file or directory.

I dont have the proper knowledge to trace this down. 
But I believe this is the same issue/error that was originally reported. & also the reason why the guide for building on this board states to use the Arduino, since something in platformio got mess up some time ago for this stm32

Eno423, 2019-11-08:
well with some more copying said directories that are reporting as non existing, I managed to get past those Errors 

Im thinking YEAH!, as it compiles for about 10 mins with just yellow warning....then finally...

it failed again...

though this time I dont think I can do anything about,

anyone have any idea?

In file included from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:33:0:
C:\Users\1996P\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\STM32ADC\src/STM32ADC.h:155:40: error: 'constexpr' needed for in-class initialization of static data member 'const 
float STM32ADC::_AverageSlope' of non-integral type [-fpermissive]
     static const float _AverageSlope = 4.3; // mV/oC   //4.0 to 4.6
                                        ^
C:\Users\1996P\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\STM32ADC\src/STM32ADC.h:156:31: error: 'constexpr' needed for in-class initialization of static data member 'const 
float STM32ADC::_V25' of non-integral type [-fpermissive]
     static const float _V25 = 1.43; //Volts //1.34 - 1.52

Eno423, 2019-11-12:
Loaded a VM with a clean build of arduino non beta ide & the 1.61 stm
built successful. 

but still cant build on my main machine using the beta/ or non beta IDE. I tried cleaning up the files, but  Im guessing I missed a few. 

only issue is my extruder isnt moving; but I am going to see If I missed something in the config

closed 2019-11-12 by Eno423

xC0000005, 2019-11-12:
If the temperature isn’t registering it may keep from extruding. Do temperatures register?

Eno423, 2019-11-19:
^^ thats probably it, 

I forgot about the cold extrude prevention. that is likley what is stopping it..

I need to finish wiring everything, as I decided to re-do 90%

I just need to now figure out the BL touch integration with the board,. but may just skip that w/ this controller (unless someone wants to spoon feed me ehh :cough: xC0000005 :cough:

xC0000005, 2019-11-19:
BL Touch works fine - with the V2, you want to wire servo to PA2, signal to PA3 (those are the two pins on the spare header) and run both sides through a level shifter.

Eno423, 2019-11-21:
> BL Touch works fine - with the V2, you want to wire servo to PA2, signal to PA3 (those are the two pins on the spare header) and run both sides through a level shifter.
> […](#)

So for the V1, is this not capable?

tvories, 2020-01-03:
I'm having build issues as well.  I see the recommended method is to use Arduino 1.6.1 but I'm not sure how to add the STM32 board to 1.6.1 before board manager.  Any suggestions?

xC0000005, 2020-01-03:
In theory the latest hardware timer updates should work across the board
without you changing the core version. In reality I've mostly been building
with my M200fixes fork and 1.6.1

tvories, 2020-01-05:
Should I be downloading libraries with Arduino to make this build work?  I don't see anything in the instructions as far as downloading libraries.

jValdron, 2020-01-18:
@tvories, if you haven't figured it out, they were talking about using STM32 1.6.1, but using Arduino 1.8.

Just tried building xC0000005's M200fixes fork, using Arduino 1.8.10, and using STM32 1.6.1. Took me a few tries to get the right configs for my machine (somewhat modded).

tvories, 2020-01-19:
@jValdron that helps a ton!  That's funny, I absolutely thought they were talking about arduino 1.6.1.  I'll retry it and see if I can get a build.