MarlinFirmware/Marlin #15623: G29 W does nothing

randellhodges, 2019-10-21:
### Description

G29 W does nothing unless UBL_DEVEL_DEBUGGING is defined

### Steps to Reproduce

Enter G29

**Expected behavior:** Useful data returned

**Actual behavior:** Nothing returned

#### Additional Information

If the entire purpose of this parameter to the command is debugging for development, this can be ignored, but some of that stuff is useful like what mesh is loaded and stuff.

ubl_G29.cpp has the entire seen G29 W wrapped in `#if ENABLED(UBL_DEVEL_DEBUGGING)` as well as the entire g29_what_command. Inside the g29_what_command, there is a block of code that is also inside an `#if ENABLED(UBL_DEVEL_DEBUGGING)`.

I'm wondering if this should be outside and executable without `UBL_DEVEL_DEBUGGING` with only that one block of code inside a check.

thinkyhead, 2019-10-21:
See if `M420 V` and `M503` provide enough information to fill in for `G29 W`.

randellhodges, 2019-10-22:
Yeah, I think those will be enough. I'll go ahead and close this. Might put a note on the marlin website for this command that it only works with ubl_devl_debugging ;)

Maybe a little room for improvement might be adding the L parameter to the M420 line that is outputed when doing a M503 S0. 

Either way, not a bug with Marlin, at most, a feature request.

closed 2019-10-22 by randellhodges