MarlinFirmware/Marlin #15624: Blank LCD on Ender 2 with SKR 1.3 board, with stock LCD

gregoryperrin, 2019-10-21:
### Hi, I have a Creality Ender 2, I just got an SKR 1.3 board, and after compiling Marlin 2.0 and putting it on the SD card, rebooting the board, the screen is lit (backlight) but blank.

I have nothing plugged on the board except a usb cable, the stock lcd screen (which works normally when i plug it on the stock creality ender 2 motherboard).

I am trying to understand where the problem comes from but it looks like it might be the configuration in Marlin?

Looking here : https://www.reddit.com/r/ender2/comments/cwxqra/pinout_for_ender2_lcd/ it looks like some people also have issue with this screen.

Thanks

### Steps to Reproduce

1. Succesfully compile marlin 2.0 with Ender 2 config
2. Copy/paste bin file to sd card
3. Put sd file in SKR 1.3 not populated except stock lcd screen plugged in EXP1

**Expected behavior:** See stuff on the screen

**Actual behavior:** Screen backlit but does not display anything

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

edwilliams16, 2019-10-22:
You'll need to do this if there's a possible problem with your configuration:
`Include a ZIP file containing your Configuration.h and Configuration_adv.h files.`

greppp, 2019-10-22:
@edwilliams16 thanks, i edited my initial post to add the zip.

greppp, 2019-10-22:
Update : someone with the same setup than me ( SKR 1.3 + Ender 2) sent me a firmware.bin file and the screen stays blank too. Which means either my board is faulty or I don't know how to upload the firmware correctly.

greppp, 2019-10-22:
Update 2 : I just tried my version and it doesn't work, blank screen again.

greppp, 2019-10-22:
semi-related, but in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/Ender-2/Configuration.h (which i used)
line 150, it says filament diameter is 3mm, which isn't true, the ender 2 uses 1.75mm.
Are those config files reliable?

greppp, 2019-10-23:
Update 3 : The user who sent me the binary file shared me his fork : https://github.com/xyyx/Marlin/tree/bugfix-2.0.x

I compiled it and now the screen works perfectly. It looks like he added a new entry in _Configuration.h_ named **#define ENDER2_DISPLAY** on line #1943. 

Then when I compile I get a lot of warnings from Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h which i guess means that's where he defined the right pins for the ENDER2_DISPLAY?

```
 #if ENABLED(ENDER2_DISPLAY)
        #define BTN_EN1      P1_18
        #define BTN_EN2      P1_20
        #define DOGLCD_CS    P1_22
        #define DOGLCD_A0    P1_21
        #define DOGLCD_SCK   P1_30
        #define DOGLCD_MOSI  P1_23
        #define FORCE_SOFT_SPI
        #define LCD_BACKLIGHT_PIN -1
      #endif
```

boelle, 2019-10-24:
@shitcreek could we extract something and make a quick (but stable) fix?

imcauseofdeath, 2019-10-30:
> **Update 3 :** The user who sent me the binary file shared me his fork : https://github.com/xyyx/Marlin/tree/bugfix-2.0.x
> 
> I compiled it and now the screen works perfectly. It looks like he added a new entry in _Configuration.h_ named **#define ENDER2_DISPLAY** on line #1943.
> 
> Then when I compile I get a lot of warnings from Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h which i guess means that's where he defined the right pins for the ENDER2_DISPLAY?
> 
> ```
>  #if ENABLED(ENDER2_DISPLAY)
>         #define BTN_EN1      P1_18
>         #define BTN_EN2      P1_20
>         #define DOGLCD_CS    P1_22
>         #define DOGLCD_A0    P1_21
>         #define DOGLCD_SCK   P1_30
>         #define DOGLCD_MOSI  P1_23
>         #define FORCE_SOFT_SPI
>         #define LCD_BACKLIGHT_PIN -1
>       #endif
> ```
> 
> I hope this helps to fix the problem in the official branch :)

Does anyone know if it is possible to use this Ender 2 fix for the SKR Mini E3 board as well?
I want to make sure it is possible before I try to track down everything this person has done to get it to work. 

Also having another strange issue on my mini board for my Ender 2. Heating up my bed also heats the hotend without stopping the hotend. But only heating the hotend, only heats the hotend?? Idk. 

Thanks!

Grrby, 2019-11-01:
Hi, After planing around, my Ender2 display works with SKR MINI E3.
I just copy solution from SKR 1.3 board and assigned proper stm32 pins.
So I added ENDER2_DISPLAY to Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_E3_DIP.h

`#if HAS_SPI_LCD
  #define BEEPER_PIN       PA15
  #define BTN_ENC          PB6

  #if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS    PB8

    #define BTN_EN1        PA9
    #define BTN_EN2        PA10

    #define LCD_PINS_ENABLE PB7
    #define LCD_PINS_D4    PB9

  //#else
    //#error "Only CR10_STOCKDISPLAY is currently supported on the BIGTREE_SKR_E3_DIP."
  #endif
 /////////////////////////////////////////////////////////////////////////////////////////
 
      #if ENABLED(ENDER2_DISPLAY)
        #define BTN_EN1      PA9
        #define BTN_EN2      PA10
        #define DOGLCD_CS    PB8
        #define DOGLCD_A0    PB9
        #define DOGLCD_SCK   PA15
        #define DOGLCD_MOSI  PB7
        #define FORCE_SOFT_SPI
        #define LCD_BACKLIGHT_PIN -1
      #endif

  //#endif

////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#endif // HAS_SPI_LCD`

I had problem with error marlin #error "Please select no more than one LCD controller option." during compilation comes from SanityCheck.h. I found a solution here [https://github.com/MarlinFirmware/Marlin/issues/5545](url) and I disabled SanityCheck.h at 
`#if COUNT_LCD_24 > 1
    #error "Please select no more than one LCD controller option."
#endif`

Maybe someone will find better and more elegant solution.

imcauseofdeath, 2019-11-01:
Regarding the 2 display error, just uncomment one of the displays so you're
only keeping one display marked as active. Right? Or is it something
different. I haven't looked at the link given with your explanation.

Thank you for the info by the way. My biggest issue is still the fact my
hotend and bed heat up when I try to preheat the bed only.

Grrby, 2019-11-01:
Actually I can't understand issue with this error because if I change  display type to default CR10, error gone, even ENDER2_DISPLAY is on the SanityCheck.h display list. When I change display in Configuration.h to Ender2 I have error again.
About your issue with preheating, I have to check with my printer.  I use fresh version of Marlin-bugfix-2.0.x and I have SKR MINI E3 DIP v1.0. What I noticed heating up is slightly slower then on original Melzi board.

imcauseofdeath, 2019-11-01:
Heating is definitely a bit slower I have noticed. But yeah, my hotend
heats up when I try to ONLY heat the bed. When I heat only the heat end,
only the heat end heats up.

It's strange and I haven't figured it out.

boelle, 2019-11-03:
@greppp still having issues?

Grrby, 2019-11-03:
For me problem is with Marlin2.0. For example, I compile everything without any warnings but when I comment MENU_HOLLOW_FRAME (what I like very much) I have a bunch of warning and display randomly shift few line up. Maybe there is some problem with my "Ender2 display solution".
What is strange, after some random time display is blank but machine still working well. I saw similar issue with SKR 1.3

I don't have any problem with heating separately bed / nozzle in any configuration.  Anyway heating up is so slow, mostly hotend.

greppp, 2019-11-04:
@boelle i haven't had time to work further on it yet, just learned how to fork Marlin today. I will try to test everything in the next two weeks.

imcauseofdeath, 2019-11-06:
> Actually I can't understand issue with this error because if I change display type to default CR10, error gone, even ENDER2_DISPLAY is on the SanityCheck.h display list. When I change display in Configuration.h to Ender2 I have error again.
> 
> ```
> /**
>  * Make sure only one display is enabled
>  */
> #if 1 < 0 \
>   + (ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER) && DISABLED(IS_RRD_SC)) \
>   + (ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) && DISABLED(IS_RRD_FG_SC)) \
>   + (ENABLED(ULTRA_LCD) && DISABLED(IS_ULTRA_LCD)) \
>   + (ENABLED(U8GLIB_SSD1306) && DISABLED(IS_U8GLIB_SSD1306)) \
>   + (ENABLED(MINIPANEL) && DISABLED(MKS_MINI_12864)) \
>   + (ENABLED(REPRAPWORLD_KEYPAD) && DISABLED(IS_RRW_KEYPAD)) \
>   + (ENABLED(EXTENSIBLE_UI) && DISABLED(IS_EXTUI)) \
>   + (ENABLED(ULTIPANEL) && DISABLED(IS_ULTIPANEL)) \
>   + ENABLED(RADDS_DISPLAY) \
>   + ENABLED(ULTIMAKERCONTROLLER) \
>   + ENABLED(PANEL_ONE) \
>   + ENABLED(G3D_PANEL) \
>   + ENABLED(RIGIDBOT_PANEL) \
>   + ENABLED(MAKEBOARD_MINI_2_LINE_DISPLAY_1602) \
>   + ENABLED(ZONESTAR_LCD) \
>   + ENABLED(RA_CONTROL_PANEL) \
>   + ENABLED(LCD_SAINSMART_I2C_1602) \
>   + ENABLED(LCD_SAINSMART_I2C_2004) \
>   + ENABLED(LCM1602) \
>   + ENABLED(LCD_I2C_PANELOLU2) \
>   + ENABLED(LCD_I2C_VIKI) \
>   + ENABLED(SAV_3DLCD) \
>   + ENABLED(FF_INTERFACEBOARD) \
>   + ENABLED(REPRAPWORLD_GRAPHICAL_LCD) \
>   + ENABLED(VIKI2) \
>   + ENABLED(miniVIKI) \
>   + ENABLED(MAKRPANEL) \
>   + ENABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
>   + ENABLED(BQ_LCD_SMART_CONTROLLER) \
>   + ENABLED(CARTESIO_UI) \
>   + ENABLED(LCD_FOR_MELZI) \
>   + ENABLED(ULTI_CONTROLLER) \
>   + ENABLED(MKS_MINI_12864) \
>   + ENABLED(ENDER2_DISPLAY) \
>   + ENABLED(FYSETC_MINI_12864_X_X) \
>   + ENABLED(FYSETC_MINI_12864_1_2) \
>   + ENABLED(FYSETC_MINI_12864_2_0) \
>   + ENABLED(FYSETC_MINI_12864_2_1) \
>   + ENABLED(CR10_STOCKDISPLAY) \
>   + ENABLED(ANET_FULL_GRAPHICS_LCD) \
>   + ENABLED(AZSMZ_12864) \
>   + ENABLED(SILVER_GATE_GLCD_CONTROLLER) \
>   + ENABLED(SAV_3DGLCD) \
>   + ENABLED(OLED_PANEL_TINYBOY2) \
>   + ENABLED(MKS_12864OLED) \
>   + ENABLED(MKS_12864OLED_SSD1306) \
>   + ENABLED(U8GLIB_SH1106_EINSTART) \
>   + ENABLED(OVERLORD_OLED) \
>   + ENABLED(DGUS_LCD) \
>   + ENABLED(MALYAN_LCD) \
>   + ENABLED(LULZBOT_TOUCH_UI) \
>   + ENABLED(FSMC_GRAPHICAL_TFT)
>   //#error "Please select no more than one LCD controller option."
>   #if COUNT_LCD_24 > 1
>     #error "Please select no more than one LCD controller option."
>   #endif
> #endif
> ```
> 
> About your issue with preheating, I have to check with my printer. I use fresh version of Marlin-bugfix-2.0.x and I have SKR MINI E3 DIP v1.0. What I noticed heating up is slightly slower then on original Melzi board.

By the way, you have done PID tune for your hotend right?

imcauseofdeath, 2019-11-06:
> Regarding the 2 display error, just uncomment one of the displays so you're only keeping one display marked as active. Right? Or is it something different. I haven't looked at the link given with [...]

Regarding my issue with preheating the bed and the hotend also starts heating

I tested tonight with a voltmeters and as soon as I preheat the bed the hotend is also (obviously) receiving 11.6V same as the heat bed side. I had to make videos of this to send to BTT. Hoping for quick replacements, my Ender 2 is down until the new one comes in. Unfortunately I had already purchased a CR10 display before finding this thread also so I doubt I'll be testing the code from above. We'll see though. 

Thanks again everyone.

Grrby, 2019-11-07:
> By the way, you have done PID tune for your hotend right?
Maybe I should but I didn't make any PID tune. Prints looks good so I don't care :-)

> I tested tonight with a voltmeters and as soon as I preheat the bed the hotend is also (obviously) receiving 11.6V same as the heat bed side.
This is obviously and maybe not means that your board is faulty. Are you using latest Marlin? Take close look at pins 39 and 40 of STM32, can be a shorts around those pins or near mosfets.

imcauseofdeath, 2019-11-08:
You for sure should do PID tune. It keeps your hotend temperature from
fluctuating too much and allows for faster heating. You can do an autotune
and store the settings.

I am running 2.0.x bugfix. I'll try to check the pins. Just hoping that
biqu stops making me go back and forth proving to them that something is
faulty.

TheAssassin, 2019-11-10:
Could anyone provide a *full*, *working* Marlin source tree with which the Ender 2 display works? Randomly combining stuff I read here once made the display work for ~10 minutes before the printer froze mid-print, now I don't get it to work at all any more. A zip would be fine, I will identify the changes and once it works set up a Git repo.
I have an SKR E3 DIP v1.0 with the STM32F103RE (not RC, RE has bigger flash size) on an Ender 2 with the stock display.

Grrby, 2019-11-10:
Welcome
[](https://github.com/Grrby/Marlin)

All files needed for SKR E3 DIP v1.0 with latest Marlin-bugfix-2.0.x 
I have STM32F103RE as well but I compiled for STM32F103RC, so it's not enough room for full option like mesh bedleveling .
Important notice. 
- use platformino core 4.0.3 ( board doesn't boot after compilation on 4.1.0).
- trash your 128MB sdcard, it is garbage. Many people report problem with it.

TheAssassin, 2019-11-10:
Thanks @Grrby. I've already set up PlatformIO to build binaries for the RE chip, I plan to send those changes upstream once I know everything works. I used to compile for RC as well, I know it works, too.

I have TMC2208 drivers.

In fact I recently updated platformio to 4.1.0, but the board does seem to flash the firmware. (Note: my board came without a bootloader, I thankfully found a GitHub issue where someone provided their bootloader which I could flash with an ST-Link v2). But I'll downgrade for testing, thanks for the hint.

I'll check your configs, if they should happen to work, I'll publish everything soon-ish.

Grrby, 2019-11-10:
To be clear, display works but not perfect. There is some field for improvement.  Sometimes there is a few top/bottom lines shift and sometimes display going to be blank (printer still working).  I don't know why. Anyway better then nothing or attaching additional display.

TheAssassin, 2019-11-10:
@Grrby okay, I've started over, adapted your configs a bit to suit my printer, and the display is alive again. I'm going to check where there's differences to my old configs.

TheAssassin, 2019-11-10:
Turns out my existing configuration worked as-is, all that broke the display is the upgrade of PlatformIO. Now this *really* shouldn't happen, but it's good we know about the issue now. Thanks @Grrby for the pointer.

Grrby, 2019-11-10:
Hmmm, I hadn't any issue with print interruption. Printer work great. Only display sometimes going crazy.

TheAssassin, 2019-11-10:
FWIW, a (1h) print finished just fine, and when I returned to the printer (~2 hours after start) the display is blank again. Weird.

Grrby, 2019-11-10:
Same issue. I have to try comment  #define BEEPER_PIN       PA15 in pins definition file
because pin PA15 is defined later as SPI Clock ( #define DOGLCD_SCK)
I see that even for default display was some problems/bugs
https://github.com/MarlinFirmware/Marlin/issues/15141

BTW Can you tell how to use STM32F103RE instead STM32F103RC ?

TheAssassin, 2019-11-10:
You basically need two extra files in `buildroot` that define the chip's layout, and a new `env` in `platformio.cfg`. I'll quickly push my changes to a new GitHub repo. Give me a few minutes please.

TheAssassin, 2019-11-10:
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...TheAssassin:fix-ender-2-display-skr-e3-dip

There you go!

TheAssassin, 2019-11-10:
Yet again, started a print, then weird 10%-of-screen-shifted-to-bottom, now after ~30 minutes screen is blank (but it's still printing from the SD card just fine).

Could this be a kind of timing/clock issue caused by us using software simulated SPI?

TheAssassin, 2019-11-11:
Here's an example of what "shifted" means on my printer:

[image]

Perhaps it helps finding the cause of the issue.

Grrby, 2019-11-11:
@TheAssassin Thank you!
It is exactly the same problem. Is some way it's good that we can reproduce this issue.   
Maybe problem caused by software SPI but from the other hand, default display is drive in the same way. My way of adapting Ender2 display is just copy/paste of solution from SKR V1.3 (LPC1768).

TheAssassin, 2019-11-12:
> Maybe problem caused by software SPI but from the other hand, default display is drive in the same way. 

Are you sure it's driven through software SPI? Can you provide a reference? (I wonder how Creality did it in the original firmware...)

I wonder whether SKR 1.3 owners have the same problem. After all, they're both ARM Cortex M3 based. I also wonder why software SPI is needed at all; STM32 chips have a hardware SPI bus. But I guess there's a reason, even if it's just that that bus is used by other functionality or incompatible wiring or anything...

Grrby, 2019-11-12:
> Are you sure it's driven through software SPI? Can you provide a reference? (I wonder how Creality did it in the original firmware...)

I'm sure.  Just take a look a STM32F103RC datasheet and/or  SKR mini E3 schematic. STM32 in LQFP64 has two hardware SPI.  One in use for r/w SD (SPI1) and second one (SPI3) is used for TMC drivers. You can read from board schematic which pins are used for LCD (TX1, RX1, SCL1, SDA1 etc)

TheAssassin, 2019-11-14:
While playing with a few settings, I noticed an SPI related section. It doesn't explicitly mention software SPI nor does it mention displays, however, as far as I can see, both the SD card and the display are controlled over software SPI on the S3 DIP.

```
/**
 * SD CARD: SPI SPEED
 *
 * Enable one of the following items for a slower SPI transfer speed.
 * This may be required to resolve "volume init" errors.
 */
#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED
```

Cutting the speed in half changes the behavior. Now, I haven't seen any shifting, instead the contrast has dropped after a while to a really low value. I'd call that progress, since I can still control the printer this way. (Edited the picture a bit to make it look somewhat like on the printer).

[image]

I'll try reducing the SPI speed for the next prints. Also I'll re-check later whether the display goes blank again. But my best guess *now* is that this problem is:

- a) a problem in the code that sets the contrast ("display goes blank")
- b) a problem with the SPI simulation in the relatively new STM32 platform for Marlin (which can seemingly be solved by reducing the clock speed)

Grrby, 2019-11-14:
All your "new" problems is a consequence of changing SD SPI speed I guess.
I've changed SPI_SPEED from 0 (fast) to 2 (slow) at u8g_com_stm32duino_swspi.cpp but without success, maybe a little bit better.

TheAssassin, 2019-11-14:
The layer height thingy is not a "new" problem, it's always been there when printing from SD.

Grrby, 2019-11-14:
I don't observe this behavior. Info on the scree following my slicer settings (I use Ideamaker 3.5 Beta and Cura 4.3.0)
BTW Looks like I've solved display problem :-)

Grrby, 2019-11-14:
**Final solution for Ender2 display on SKR Mini E3**
Forgot about everything from my previous post. All what you need is to add new pins definition to pins_BIGTREE_SKR_E3_DIP.h (or pins_BIGTREE_SKR_MINI_E3.h if have this one) 

`  #if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS    PB8

    #define BTN_EN1        PA9
    #define BTN_EN2        PA10

    #define LCD_PINS_ENABLE PB7
    #define LCD_PINS_D4    PB9

  //#else
    //#error "Only CR10_STOCKDISPLAY is currently supported on the BIGTREE_SKR_E3_DIP."
  #endif
 /////////////////////////////////////////////////////////////////////////////////////////
 /**ENDER2 MINIDISPLAY
 *                   _____
 *               5V | · · | GND
 *      (MOSI) PB7  | · · | PB8  (LCD_RS)
 *    (LCD_A0) PB9  | · · | PA10 (BTN_EN2)
 *            RESET | · · | PA9  (BTN_EN1)
 *   (BTN_ENC) PB6  | · · | PA15 (SCK)
 *                   -----
 *                    EXP1
 */
 
      #if ENABLED(MKS_MINI_12864)
        #define BTN_EN1      PA9
        #define BTN_EN2      PA10
        #define DOGLCD_CS    PB8
        #define DOGLCD_A0    PB9
        #define DOGLCD_SCK   PA15
        #define DOGLCD_MOSI  PB7
        #define FORCE_SOFT_SPI
        #define LCD_BACKLIGHT_PIN -1
      #endif

  //#endif
`

Next at Configuration.h, 
comment //#define CR10_STOCKDISPLAY 
and uncomment 
#define MKS_MINI_12864

That's all. Display works perfect.

TheAssassin, 2019-11-14:
Interesting. I can't see which pin caused the trouble (I suspect it's `CD_RS`, but that's just a guess right now). Thanks. I'll try it on vanilla Marlin, and if it works will send a PR.

Grrby, 2019-11-14:
I've tested this solution all day and no display issue. On Marlin 2.0 and Melzi board MKS_MINI_12864 is used. So decided to test is, I've just added pins decision which I already defined before.
I'm new to Github and I'm learning how to use is, so please do whatever is helpful for other user.  Ender2 is a bit forgotten in Ender3 era but I'm sure lot of people still use it. Upgrading the board is one of the best upgrade. I have both of those printers but I like Ender 2 very much and this is my workhorse.

TheAssassin, 2019-11-14:
Yeah, I love my Ender 2, it's my portable printer. If my bed wasn't so warped already in the corner on the upper left, I'd use it much more often.
Defining a default contrast in `Configuration.h` removes that contrast option. I like `160` for it.

TheAssassin, 2019-11-14:
You have the SKR E3 Mini, right? I'm adding the pins definition for this board as well, and I need a tester. When I prepare the PR, you can just clone that branch then, put your `Configuration.h` and `Configuration_adv.h` in, build and test. Would you like to do that?

Grrby, 2019-11-14:
Sure. Yes, I have SKR Mini E3 V1.2. Still in the box :-). My plan is to use on my Ender3 but I can do some test. 
BTW  BIGTREETECH just started to sell TFT35 E3 V3.0, super cool "dual" display for ender3. I don't know if is allowed to put link here to Ali.

TheAssassin, 2019-11-14:
Sure, why not?

Grrby, 2019-11-14:
Micro SD is an nightmare. I have original encloser on my Ender2 and SKR board is bit shorter then Melzi, so now I need tweezers to remove SD. I'm going to cutout a base for better access.    

[](https://pl.aliexpress.com/item/4000349509983.html)
I just don't understand what means Limited Price and Original Price (price is different ofc but I can't realize is there any difference in product)

TheAssassin, 2019-11-14:
You probably saw that I already laser cut a new case out of HDF/Hardboard at a local FabLab. I have tons of old small USB2 pen drives, which I want to use there.

Grrby, 2019-11-14:
You integrated video projector as well :-)

TheAssassin, 2019-11-14:
Not quite, just a 12V on/off switch. The flash just makes it look like this.

[image]

The stock Ender 2 board had two pins for a power switch, now I had to wire the main power to this front switch. Pretty annoying, actually.

Grrby, 2019-11-14:
I use modified server's power supply and front switch on the printer is just a reset for, connected to pins on SPI header (RST-GND).

TheAssassin, 2019-11-14:
> connected to pins on SPI header (RST-GND).

Brilliant idea, I was already thinking how to implement a reset button on the front. I didn't think of simply using the SPI header for this.

Grrby, 2019-11-14:
Original power supply is a cheap and noisy. I never trusted it. I always power off on power main switch.  I found high quality HP server power supply and after some mods it's a perfect replacement.

Back to display. From time to time, I see some artifacts on the screen. Just for a fraction of second but later nothing happening wrong. Probably a those moments display shifting or going blank on my previous solution.

TheAssassin, 2019-11-14:
> Original power supply is a cheap and noisy. I never trusted it. I always power off on power main switch. I found high quality HP server power supply and after some mods it's a perfect replacement.

Mind to share some details (e.g., model number)? I'm thinking about using a Mean Well LRS-200-12 (passive, rated 50W less than the original but I have doubts the original draws even 200W).

I haven't seen any artifacts yet, however I don't look at the display all the time.

Grrby, 2019-11-15:
My little beast is a HP DPS-800GB. You can find it almost for nothing or you can use something similar. Just check if it possible to mod it. Server supplies are very high quality but some of them are almost impossible to turn to stand alone mode because of very complicated driving.
 
https://pa0fri.home.xs4all.nl/Diversen/DPS-800GB%20A%20Server%20sypply/DPS-800GB%20A%20PSU%20eng.htm
https://krusic22.com/2017/04/01/hp-dps-800gb-a-12v-psu-modding/

ManuelMcLure, 2019-11-15:
I use an HP DPS-600PB for my “was once, long ago, an Anet A8” printer. It’s dead easy to make work, supports PS_ON and provides 47 amps at 12.5V. The only thing I don’t like about it is that the fan is always on even if it’s turned off via PS_ON - otherwise I’d power my RPi from the 5VSB rail instead of using the wall-wart.

Grrby, 2019-11-15:
I had HP DPS-600PB as well (from Prolinat DL380G3 I guess), Only one jumper is needed to turning it on, but yes, fan is quite noisy.

Grrby, 2019-11-21:
Smart and simple stand-off for mounting SKR Mini E3 V1.0/V1.2 /DIP 1.0 and Creality 1.1.5 Silent board on the ENDER 2.
https://www.thingiverse.com/thing:3965485

TheAssassin, 2019-11-22:
That guide suggests to use `MINIPANEL` for the display...

brenca, 2019-11-25:
Thanks @TheAssassin for the PR! I bought an `SKR Mini E3 v1.2` for my `Ender 2`, and been struggling with the display, and the comments here helped me to get things kinda going, and now I hope that the glitches will go away.
The pinout didn't work for me (blank screen) but I figured out what was causing that, I'll post a PR with a fix shortly.

[4 more replies]

closed 2019-12-07 by boelle