MarlinFirmware/Marlin #15626: [FR] (add M106/M107 in marlin to control a servo)

loler-later, 2019-10-22:
I am a college student and my team is working on a project that uses marlin firmware to control steppers and servos. I find out that it is set to use M280P0S in Marlin to control servos, but the software I use can only generate M106/M107 or M3/M5 to turn on/off something. I am looking for someone to help me use M106 to set the servo to 30 degrees and M107 to set it to 0 degrees.

psavva, 2019-10-22:
This what you looking for?
http://marlinfw.org/docs/gcode/M280.html

Grogyan, 2019-10-22:
Those Gcodes are already assigned to other operations.
Recommend checking into your software/slicer as M280 is the correct gcode for controlling servos

shitcreek, 2019-10-24:
What sort of software are you using that can only generate M106/107 and M3 - M5?
If you are stuck with that software, a simple hack would be remove M106/M107 from Marlin and change any mentioning of M280 to M106. Or copy the M280's codes over into M106.

boelle, 2019-11-03:
since this is not a bug with marlin going to close this one

we can reopen if there is a bug as such

closed 2019-11-03 by boelle