MarlinFirmware/Marlin #15628: Err: Too far! Anet A8

aamerF, 2019-10-22:
greetings,
 just a quick question :
after updating my stock A8 to marlin 1.1 with the correct configuration profile , after auto home the display shows: X -33 Y -10 Z 0 , and The home position for my printer is not above the bed  . so I am trying to adjust the X and Y offset . I'm not sure if I am doing this correctly but I use the menu to move the the correct corner of the bed and then select "Set home offsets" but I get an error: Err: Too far!

reloxx13, 2019-10-22:
Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Or use discord: https://discord.gg/n5NJ59y

boelle, 2019-10-22:
as @reloxx13 said ask questions elsewhere, the list is for bugs and feature resquestions only

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-10-22 by boelle

aamerF, 2019-10-22:
if (ask questions elsewhere), y other user asking in same (issue)section ,, and replay to them . what i ask is support i need because my printer stop working and i can't use , thank u very much to u r replay

boelle, 2019-10-22:
this list is not for questions and support, bugs only, and this is not only you
this "rule" is for everyone

aamerF, 2019-10-22:
sorry for that . i don't know about that rules here , plz if u know any think about my issue can u support me plz  ,in this time ,, i visit all link @reloxx13 provided to me , no replay i get or support ,,
thank u for u r under stand