MarlinFirmware/Marlin #15629: [FR] Marlin 1.1.x CLCD Korean support

portalab, 2019-10-22:
### Description
I think it is cool if my delta print supports korean CLCD interface.
But not support on 1.1x 
So I will try making korean interface for fun.

Language support list : http://marlinfw.org/docs/development/lcd_language.html

### Steps to Reproduce
1. Makes font with a few characters
2. translate as a code
3. Test
?

#### Additional Information
http://marlinfw.org/docs/development/lcd_language.html#mappers

boelle, 2019-10-22:
use 2.0.x instead, 1.1.x is not updated much anymore

portalab, 2019-10-23:
> use 2.0.x instead, 1.1.x is not updated much anymore

Okay, but My friends and me still use 1.1.9 release, so it will be great if it support  on 1.1.x  :)

boelle, 2019-10-23:
upgrade to 2.0.x, there is not much support on 1,1.x

shitcreek, 2019-10-24:
@portalab if you're willing to do all the legwork, I will gladly create a PR for it :)

closed 2020-06-30 by github-actions[bot]