MarlinFirmware/Marlin #15630: [BUG] G26 not printing mesh on entire bed

randellhodges, 2019-10-22:
### Description

UBL is active and a mesh is loaded. When I execute G26 (no arguments) it just prints the circle over and over on top of itself without ever moving to another location.

SKR 1.3

### Steps to Reproduce

1. Use UBL and load a mesh
2. G26

**Expected behavior:** It to move around the bed printing the pattern.

**Actual behavior:** Just prints the circles on top of itself never moving to a new location

#### Additional Information

Configuration:
https://github.com/randellhodges/Marlin/tree/9905adc79cb9844cea3dd67a3a94ddeef79859ef/Marlin

shitcreek, 2019-10-22:
What does your mesh look like? How and did it successfully load?

randellhodges, 2019-10-22:
```
Send: G29 T1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: -0.062	-0.062	-0.060	-0.063	-0.017	0.027	0.057
Recv: -0.021	-0.028	-0.035	-0.037	-0.011	0.049	0.076
Recv: 0.122	0.074	0.026	-0.005	-0.003	0.038	0.017
Recv: 0.127	0.089	0.051	0.012	0.006	0.011	0.005
Recv: 0.129	0.087	0.046	-0.002	-0.013	-0.019	-0.044
Recv: 0.109	0.066	0.023	-0.014	-0.032	-0.012	-0.032
Recv: -0.020	-0.034	-0.048	-0.046	-0.023	0.029	0.068
```

```
Send: M503
...
Recv: echo:Unified Bed Leveling:
Recv: echo:  M420 S1 Z1.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: 
Recv: Active Mesh Slot: 0
Recv: EEPROM can hold 16 meshes.
...
```

Roxy-3D, 2019-10-22:
My guess is something got changed without proper testing...
Can you pull down a snap shot from 3 months ago and see if that works.

If it does...   I'll go through the various commits and figure out what broke the code.

randellhodges, 2019-10-22:
I can do that. It might be a day or so, but I'll try it and give it a go. Any specific-ish date you had in mind?

Roxy-3D, 2019-10-22:
No...   Just pick a snap shot about 3 months ago...    And if that works for you, almost for sure something got merged that shouldn't have.      And I'll start looking at the differences and see what changed.   

(Probably not that much changed.  G26 has been stable for 18 months...  So it shouldn't be that hard to find the bad code if that is what we are fighting.)

Jimbo145, 2019-10-23:
This is caused by the issue below, just haven't had the time to submit a PR

https://github.com/MarlinFirmware/Marlin/issues/15592

Roxy-3D, 2019-10-23:
@randellhodges    

Can you give the current v2.0.0 BugFix a try?    I think I have it patched based on what @Jimbo145  said up above.

randellhodges, 2019-10-23:
@Roxy-3D I'll give it a try tonight. Funny, I made the changes based on @Jimbo145, but hadn't tried it yet. I'll report back later tonight.

dongl1, 2019-10-23:
I can tell you that the fix mentioned by jimbo145 works, as I was having the same issue and this fixed it.

Roxy-3D, 2019-10-23:
The problem with what @Jimbo145  reported is...  There are three inline functions with the const xy_int8_t structure.   His note only mentioned two of functions.   I went ahead and changed the definition for the third one also.    

But I have no way to test things at the moment.   So it would be good to get confirmation from somebody that what I did is appropriate:
```
  struct FlagBits {
    typename IF<(W>8), uint16_t, uint8_t>::type bits[H];
    void fill()                                   { memset(bits, 0xFF, sizeof(bits)); }
    void reset()                                  { memset(bits, 0x00, sizeof(bits)); }
    void unmark(const uint8_t x, const uint8_t y) { CBI(bits[y], x); }
    void mark(const uint8_t x, const uint8_t y)   { SBI(bits[y], x); }
    bool marked(const uint8_t x, const uint8_t y) { return TEST(bits[y], x); }
    inline void unmark(const xy_int8_t &xy)       { unmark(xy.x, xy.y); }
    inline void mark(const xy_int8_t &xy)         { mark(xy.x, xy.y); }
    inline bool marked(const xy_int8_t &xy)       { return marked(xy.x, xy.y); }
  };
```

dongl1, 2019-10-23:
Hey Roxy I changed all three as well and it works as it should.

randellhodges, 2019-10-23:
Thanks @Jimbo145 and @Roxy-3D, it looks like I'm getting circles everywhere now!

Fixed with 3642a12fc170ffdee8e5db5d78684f82914edc41

closed 2019-10-23 by randellhodges

Roxy-3D, 2019-10-23:
Incidentally....  If we had gone down the suggested path of loading a snap shot from 90 days ago...   We would have been able to confirm that worked.  We would have been able to look at changes from that point forward and we would have found:

[image]