MarlinFirmware/Marlin #15632: [FR] TMC2225 stepper driver support

Hedda, 2019-10-22:
[Feature Request] Support for Trinamic TMC2225 stepper drivers

https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2225_Datasheet_V109.pdf

"_The TMC2225 is a Step/Dir Driver for Two-Phase Bipolar Stepper Motors up to 2A peak - StealthChop for Quiet Movement - UART Interface Option._"

As I understand it; TMC2202, TMC2208, TMC2220, TMC2224 and TMC2225 all belong to the same stepper driver family from Trinamic?; "The TMC22xx family". See:

https://www.digikey.com/en/datasheets/trinamic-motion-control-gmbh/trinamic-motion-control-gmbh-tmc220x_tmc222x_datasheet

@BigTreeTech TMC2225 v1.0 drivers are supposedly available from BIGTREETECH now?
https://www.reddit.com/r/BIGTREETECH/comments/dlgfcw/new_driverbigtreetech_tmc2225/

teemuatlut, 2019-10-22:
We'll see what their actual status is first. The IC isn't listed anywhere on Trinamic's site and all a google search finds is the datasheet on this very FR.

LinFor, 2019-10-23:
Interesting thing found in datasheet:

> Internal Step Pulse Generator
> Some applications do not require a precisely co-ordinate motion – the motor just is required to move
> for a certain time and at a certain velocity. The TMC22xx comes with an internal pulse generator for
> these applications: Just provide the velocity via UART interface to move the motor. The velocity sign
> automatically controls the direction of the motion. However, the pulse generator does not integrate a
> ramping function. Motion at higher velocities will require ramping up and ramping down the velocity
> value via software.
> STEP/DIR mode and internal pulse generator mode can be mixed in an application!

Maybe someday we can use this feature

hackebike, 2019-10-24:
[datasheet](https://webcache.googleusercontent.com/search?q=cache%3A8j-gr68FOjEJ%3Ahttps%3A%2F%2Fwww.trinamic.com%2Ffileadmin%2Fassets%2FProducts%2FICs_Documents%2FTMC2225_Datasheet_V109.pdf%20&cd=1&hl=en&ct=clnk&gl=us&fbclid=IwAR1BW3Sr8bYl4IrF8EPY_zZLrqqGgo4S9mK1Y3iv9TDxq9rDjWTbfSOZNSg)

bagobor, 2019-10-24:
@teemuatlut @hackebike  @Hedda @LinFor I've found it here https://www.ebay.com/itm/6Pcs-TMC2225-V1-0-Stepper-Motor-Driver-Stepstick-with-Heatsink-For-SKR-V1-3-MKS/274065785779 going to order now. 
Have spare BG SKR v 1.1 board as well so I'll be able to perform few tests.

hackebike, 2019-10-25:
they are on sale at BTT store on [aliexpress](https://bit.ly/2ML5JJw) with some more information

Hedda, 2019-10-28:
Direct link https://www.aliexpress.com/item/4000296898203.html

bagobor, 2019-10-29:
@Hedda  looks like I'll get it faster anyway ;P

bagobor, 2019-11-08:
Was order was delivered today x6.

bagobor, 2019-11-08:
It looks like drop-in replacement for 2208 drivers. 
https://wiki.fysetc.com/TMC2225/

bagobor, 2019-11-09:
https://github.com/teemuatlut/TMCStepper/issues/90

bagobor, 2019-11-10:
@Hedda according to https://github.com/bigtreetech/BIGTREETECH-TMC2225-V1.0/blob/master/TMC2225%20V1.0%20manual.pdf UART mode is compatible with 2208 drivers.

Hedda, 2019-11-15:
Not listed here yet though https://github.com/bigtreetech/TMCStepper

bagobor, 2019-11-15:
@Hedda I've ordered SCR PRO new board to test it. It should arrive tomorrow. And then will create PRs to marlin and TMCStepper repos.

bagobor, 2019-11-17:
SRC PRO came broken (thanks Amazon!) I've ordered a replacement ...

dwillmore, 2020-01-09:
Datasheet link went 404.

bagobor, 2020-02-26:
@dwillmore https://www.digikey.com/en/datasheets/trinamic-motion-control-gmbh-tmc220x_tmc222x_datasheet

bagobor, 2020-02-27:
Looks like this product is not supported so no point to added its support here.

GadgetAngel, 2020-03-01:
> Datasheet link went 404.

They now have it listed under TMC2208, here is the link:
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf

closed 2020-06-30 by github-actions[bot]

zapl, 2020-07-11:
There is a datasheet out there for the TMC2225, it's a slightly reworked version of that for the 22xx family (2202,2208,2224). TMC2225 is a TMC2224 in HTSSOP28 package rather than the smaller QFN28. That's it.

There aren't any (marlin) relevant changes from 2208 to 2224/2225 it seems. The 2224/2225 has different config pins and would run at 32 microsteps if configured the same MS1/MS2 as 2208. If also has a SPREAD pin to flip the logic of `en_spread_ cycle`, i.e. you can default to spread cycle without UART or OTP.
The only thing that might become relevant for marlin is a change in the meaning of `IOIN` (input pin states) register bits, those changed. E.g.  bit 2:
MS1 (TMC220x), SPREAD (TMC222x). I've just glanced at the code, they are currently merely printed and not evaluated in any way.

GadgetAngel, 2020-07-21:
BIGTREETECH has a TMC2226 where is the data sheet on this device?

hackebike, 2020-07-22:
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2226_Datasheet_V106.pdf