MarlinFirmware/Marlin #15633: [FR] (feature request title)

TOGtribeofgamers, 2019-10-22:
Hi, I have an ender 5 with Marlin 2.0, what value should I set to report the steps as in the picture?
I apologize for my incorrect English but I speak Italian.
[image]
[image]

reloxx13, 2019-10-22:
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.
> 
> Or use discord: https://discord.gg/n5NJ59y

Or join Creality/Ender groups on FB.

and read marlin gcode docs.

TOGtribeofgamers, 2019-10-22:
ok thanks. delete post

closed 2019-10-22 by TOGtribeofgamers