MarlinFirmware/Marlin #15634: TEMP_SENSOR_1_AS_REDUNDANT #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

DroneMang, 2019-10-22:
Hi,

I am working on trying dual thermistors for 1 hotend.  If I set the below I get the error.  Does this assume both thermistors are using the same sensor, and that temp sensor 1 gets the same value of 5 in this case because it is set for temp sensor 0?

If I set:

#define TEMP_SENSOR_1_AS_REDUNDANT

and set:

#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5

I get this error during compiling:
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

Thank you,
DroneMang

shitcreek, 2019-10-22:
Which version of Marlin are you using? In Marlin 2.0, this error is produced by `SanityCheck.h` when `TEMP_SENSOR_1_AS_REDUNDANT` is disabled:
`#elif TEMP_SENSOR_1 != 0 && DISABLED(TEMP_SENSOR_1_AS_REDUNDANT)
  #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."`

boelle, 2019-10-22:
and also i think config files would help?
and also give direct link to where you got marlin from, just so we know what we are looking at

psavva, 2019-10-22:
I'm going to suggest you try work the lastest Marlin 2.0. Looking so the code, I cannot tell that you'd be able to simulate the same error.

The other alternative is that something is mistyped in your conf files .

Please upload

boelle, 2019-10-22:
@psavva who should upload what? we dont know if there is something mistyped as there are no config files

psavva, 2019-10-22:
The person reporting the issue ;)

boelle, 2019-10-22:
what should he upload that have not allready been asked for?

DroneMang, 2019-10-22:
Here are the config files:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3758483/config.zip)

I downloaded this last night from github.
[Marlin-2.0--3.zip](https://github.com/MarlinFirmware/Marlin/files/3758486/Marlin-2.0--3.zip)

GMagician, 2019-10-22:
I tested attached cnf and compile fine (but using vscode)

boelle, 2019-10-22:
what version of ide? 1.9.0 beta is recommended.... or vscode+platform.io

GMagician, 2019-10-22:
Compiled correctly with vscode and ide 1.8.10

DroneMang, 2019-10-22:
With temp 1 set to 5 and not 0?

boelle, 2019-10-22:
prob as the config files are uploaded, so if the configfiles you attached is with temp 1 set to 0 then yeah

you should always upload files with correct settings even if they dont compile

DroneMang, 2019-10-22:
Here are the config files that produce the error.  I'll upload with the correct settings in the future.  Thank you.

[config with error.zip](https://github.com/MarlinFirmware/Marlin/files/3759016/config.with.error.zip)

GMagician, 2019-10-23:
It seems that to enable such function you have to write `#define TEMP_SENSOR_1_AS_REDUNDANT true`
but this is in contrast to what is written in all configuration files...

GMagician, 2019-10-23:
@DroneMang please test my PR nr #15650

DroneMang, 2019-10-25:
This compiles now with no errors (except the DOGM one which I'm assuming will get fixed soon).

Question, is there a way to view the temperature outputs of each thermistor in the terminal?  I would like to do some tests.

As it is now, the logic says to kill the print if the temperature difference is too large.  My friend pointed out that it could also be used in a redundant type way, where if one thermistor fails, a flag could be thrown but the print could continue with the 1 good thermistor.  But then how do you with confidence decide which is the good one and which to trust?

I prefer the current logic, but perhaps another option could be in the config file to allow the user to decide.

boelle, 2019-10-30:
@GMagician when will the PR be sent to 2.0.x ? if allready done what is the PR number?

GMagician, 2019-10-31:
Already been merged as I know it's #15650

boelle, 2019-10-31:
you are right

will risk my neck and close this one then

we can always reopen if PR https://github.com/MarlinFirmware/Marlin/pull/15650 did not solve it

closed 2019-10-31 by boelle

DroneMang, 2019-10-31:
Just tested this last night. It works, but when the hotend is heating up it
must heat the sensors at different rates as it emergency shut down and
disconnected. I then reconnected and had it heat up to pla temps and it was
fine. I can just make the temp difference 15 instead of 10, but maybe there
can be a feature to wait until at temp before comparing them?