MarlinFirmware/Marlin #15638: [BUG] SKR v1.1 Bed leveling filed (bugfix-2.0.x)

weatherwarlock, 2019-10-22:
### Description
Delta with chimera with one nozzle
board: SKR v1.1 with TMC2130 SPI

### Steps to Reproduce
M502
M500
G28
G33 P7 V2

#### Additional Information
The log of a couple of attempts in the archive. Last attempt with settings from the archive
[BedLevelingError.zip](https://github.com/MarlinFirmware/Marlin/files/3758453/BedLevelingError.zip)

weatherwarlock, 2019-10-22:
Probe as removables
https://www.thingiverse.com/thing:2492071

shitcreek, 2019-10-22:
If you are using a removable probe, you should use `FIX_MOUNTED_PROBE` instead of `Z_PROBE_ALLEN_KEY`

weatherwarlock, 2019-10-23:
my choice was based on a description(http://marlinfw.org/docs/configuration/probes.html):
> Z_PROBE_ALLEN_KEY This is a popular solution on deltas. A spare Allen key is used with an endstop switch to make a probe that’s deployed and stowed by turning the key 90 degrees. **You can either deploy and stow the key manually** or configure movements that bump the key against some fixed point. Options for this type of probe are included in the delta example configurations that come with Marlin.

And earlier like it was just another description on reprap

weatherwarlock, 2019-10-23:
I'm sorry, the contact was unstable
when I checked m119 - everything was fine, but when calibrating - it left

closed 2019-10-23 by weatherwarlock