MarlinFirmware/Marlin #15639: [FR] Switchless homing

Zol-Tank, 2019-10-22:
### Description

Switchless homing. I'd would be nice if I could manually move the axis to their home position, tell Marlin "They're at home position", and have it simply accept those positions as home. On custom printers, it would be nice not having to worry about endstop switches.

hackebike, 2019-10-23:
isn't that sensorless homing? I have a custom printer and I set some fix ends that make the home position, than using sensorless homing, the motor stops at the fix position.

Zol-Tank, 2019-10-23:
Sensorless homing uses feedback from the steppers stalling to set home, requiring stepper drives capable of sensorless homing running in UART or SPI, plus having some robust for your axis to run into at the home position. What I'm talking about doesn't need any of that. It's simply telling Marlin "Your axis are at their home position right now".

hackebike, 2019-10-23:
I see, so is a software config.

Zol-Tank, 2019-10-23:
Yeah, basically. You move the axis by hand to their home position, and then you tell Marlin "They're at their home position". It just simplifies the hardware a bit, which is nice for custom or heavily modified printers.

Zol-Tank, 2019-10-23:
Yes, they do, and they can even damage the drivers if you move them fast enough. However, I figure switchless homing like this would probably only be used by fairly experienced people who'd know better than to go zooming their axis around.

LinFor, 2019-10-24:
G92 X0 Y0 Z0  is not a solution to your problem?

Zol-Tank, 2019-10-24:
> "G92 X0 Y0 Z0" is not a solution to your problem?
> It can be added in "custom commands" menu if you need use it from LCD

I had no idea that was even a thing. That's great! So if I have a BLtouch, I should be able to do this to home the axis then level the bed, correct?
G92 X0 Y0
G28 Z
G29

LinFor, 2019-10-24:
> So if I have a BLtouch, I should be able to do this to home the axis then level the bed, correct?

I didn't check this but probably yes

thinkyhead, 2020-06-24:
The `MARLIN_DEV_MODE` option gives you `G28 S` for "simulated homing" with no movement. Use at your own risk.

closed 2020-06-24 by thinkyhead