MarlinFirmware/Marlin #15641: Z position wrong when tool changes after G29

SwiftNick, 2019-10-23:
After homing and bed levelling, hot end is 5mm above bed and display shows z = 5
When tool selected, Z position on screen changes to 1 but the z axis does not move.
Print starts 4mm above the bed.

The same issue occurs using Marlin-bugfix-2.0.x.
Older Marlin-bugfix-1.1.x works just fine.

**Configuration:**
  MOTHERBOARD:  BOARD_K8400
  EXTRUDERS:  2
  Z_PROBE_SERVO_NR:  0  (using a BLTouch probe)
  AUTO_BED_LEVELING_LINEAR

**Steps to reproduce**
Issue the following commands via terminal.
G28     
G29
T0

I you don't do the bed levelling the Z position is fine and does not change during a tool change.

Evgeny-SPB, 2019-10-23:
Hello! Your problem seems very similar to the one described in https://github.com/MarlinFirmware/Marlin/issues/15610
You can try to reproduce steps 
G28 
G29 
M114 D
The last command with D modifier gives coordinates before applying rotation with AUTO_BED_LEVELING_LINEAR and after.  
(Remember that for M114D to work you need to #define M114_DETAIL in Configuration_adv.h)

You have dual extruded setup. I have dual Z setup. In my case the problem was with planar bed leveling (AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT), which both use the same function for rotation.

You could also try to use other bed leveling methods (maybe unified bed leveling), as I understand, they use other function for applying bed levelling.

SwiftNick, 2019-10-23:
Your suggested fix in `vector_3::apply_rotation` seems to have resolved this one.
Nick

Evgeny-SPB, 2019-10-23:
I’v spent about 20 hours on this bug. I’m glad that my solution helped you 👍

boelle, 2019-10-24:
so this one is allready solved?

Evgeny-SPB, 2019-10-24:
Pull request is generated regarding this issue. It is not yet merged though.

boelle, 2019-10-24:
oki, will close then

we can always reopen

closed 2019-10-24 by boelle