MarlinFirmware/Marlin #15643: Position offset is not applied when change from T1 to T0

SwiftNick, 2019-10-23:
### Description
When changing tools, the bed should move based on the HOTEND_OFFSET values. 
When changing from T0 to T1 the bed moves correctly.
When changing from T1 to T0 the bed does not move.

### Steps to Reproduce
Issue these commands
T0
G28
T1
T0

**Expected behavior:** [What you expect to happen]
When T1 selected the bed should move to HOTEND_OFFSET
When T0 selected the bed should move back.

**Actual behavior:** [What actually happens]
The bed moves when T1 is selected, but does not move back when T0 is selected.

#### Additional Information
#define EXTRUDERS 2
#define HOTEND_OFFSET_X { 0.0, 35.00 } // (mm) relative X-offset for each nozzle
#define HOTEND_OFFSET_Y { 0.0, 0.00 }  // (mm) relative Y-offset for each nozzle
#define HOTEND_OFFSET_Z { 0.0, 0.00 }  // (mm) relative Z-offset for each nozzle

SwiftNick, 2019-10-23:
Tool change is not taking the current offset position into account.
A working fix in `tool_change.cpp` is to change line 909 from
`xyz_pos_t diff = hotend_offset[new_tool];`
to
`xyz_pos_t diff = hotend_offset[new_tool] - hotend_offset[old_tool];`

boelle, 2019-10-24:
maybe submit a PR so others can benefit?

SwiftNick, 2019-10-24:
Have you done a pr for this one?
Was reading up on GitHub not used it much before!

boelle, 2019-10-24:
no i have not

closed 2019-10-24 by SwiftNick