MarlinFirmware/Marlin #15648: When T1 is active M107 shuts down the wrong cooling fan

SwiftNick, 2019-10-23:
### Description

Printer has 2 extruders and 2 sets of cooling fans.
Fan P0 is print cooling fan
Fan P1 is extruder fan
When T0 is active, M107 correctly shuts down the print cooling fan P0 
When T1 is active, M107 shuts down the extruder fan P1. This is a bad thing!

### Steps to Reproduce
```
M106 P0 S200  turn on print cooling fans
M106 P1 S200  turn on extruder fans

T0            select tool 0 
M107          correctly turns off print cooling 

T1            select tool 1
M107          Now turns off extruder extruder fan 
```
**Expected behavior:** 
Extruder fan should not be shut down by M107 command with no P value.

AnHardt, 2019-10-23:
If you want to shut down a specific fan use the P parameter.
Without P M107 shuts down the fan of the "active tool" - exactly as expected.
M106 behaves exactly the same.

SwiftNick, 2019-10-23:
I agree but the slicing software such as cura doesn't specify a P value for M107 which means I would have to search and replace every M107 command in all print jobs.

The GCode spec says that M107 without a P value should shut down the cooling fan.
[Marlin Gcode Spec for M107](http://marlinfw.org/docs/gcode/M107.html)

It works fine when i'm using T0 but when I use T1 it changes the value of the print cooling fan, which is wrong.

SwiftNick, 2019-10-23:
I think the ```_ALT_P``` default return value in ```const uint8_t p = parser.byteval('P', _ALT_P);```
called by ```GcodeSuite::M106()``` and ```GcodeSuite::M107()``` should always be 0 as the print cooling fan doesn't change when you switch extruders.

AnHardt, 2019-10-23:
You connected the fan, cooling the cold end, to a connector ought to be a part cooling fan. Set the P1_pin to -1 and use a AUTO_FAN with the overwritten pin_number.

SwiftNick, 2019-10-23:
You are correct I did, as I run the extruder fans at 80% to keep the noise down.
Autofan does work much better but I can't get it to support soft PWM.

AnHardt, 2019-10-23:
So the bug is not the behavior of M106/M107,
but the inability to use SOFT_PWM on the AUTO_FAN_PIN.

SwiftNick, 2019-10-23:
Ha ha more of a feature request!
I think there is still a bug as v2.x doesn't follow the Marlin GCODE spec for M106 & M107.

boelle, 2019-10-24:
do you know a fix for this one also?

SwiftNick, 2019-10-24:
I do have a fix.
change Line 35 in M106_M107.cpp
from ```#define _ALT_P _MIN(active_extruder, FAN_COUNT - 1)```
to ```#define _ALT_P 0```

With no P parameter specified M107/M106 should always reference the print cooling fan, which is usually FAN0
If a machine has multiple print cooling fans then the P parameter will override this.

The behaviour in v2 has changed.
It now controls the print cooling fan of the active extruder when no P parameter is present.
Which makes sense, but is not the behaviour defined in the GCODE spec.
So I think the default fan should be 0 which is how v1.1.x behaves.

boelle, 2019-10-24:
could you submit a PR with the fix?

SwiftNick, 2019-10-24:
Will do

boelle, 2019-10-24:
awesome, things are fixed faster that way

SwiftNick, 2019-10-24:
which branch should I use? bugfix-2.0.x or dev-2.1.x?

boelle, 2019-10-24:
2.0.x

AnHardt, 2019-10-24:
You will break the tool-changers trusting in they don't have to care about what part cooling fan to switch. They send M106/M107 without a parameter and always find the right fan.

Don't destroy this for a abuse of a **part**-cooling-fan.

Likely it needs a much advanced part-cooling-configuration to satisfy all possible valid nozzle/tool/part-cooling-fan combinations.

But your use case is not even valid.

SwiftNick, 2019-10-24:
I guess this is an issue with the massive variety printers.
A warning to those upgrading from v1.1, the behaviour has changed and it causes damage to a dual extruder machine with a single print cooling fan. As the extruder fan gets switched off during the print.

AnHardt, 2019-10-24:
A cold-end-fan is not a part-cooling-fan!
In crappy (historically grown :-) ) Marlin terminology: a FAN is not a AUTO_FAN.

SwiftNick, 2019-10-24:
So FAN_PIN and FAN1_PIN are for fans that cool the part, and are addressed using M106 P0 and M106 P1

E0_AUTO_FAN_PIN & E1_AUTO_FAN_PIN are cold end cooling fans driven by the firmware only.
I shouldn't be using FAN1_PIN to control my cold end cooling fan.
So I need to find out why AUTO_FAN PWM doesn't work on pin2 which is a HW PWM pin.
I'll close this and move on :-)

closed 2019-10-24 by SwiftNick

AnHardt, 2019-10-24:
https://github.com/MarlinFirmware/Marlin/issues/15648#issuecomment-545665938