MarlinFirmware/Marlin #15649: [BUG] (or feature?) "MINIS" string in gcode file name causes printer to well... print miniatures.

salacpavel, 2019-10-23:
Now here is the odd one. First I thought I hit some weird bug in Cura. So I have resliced and saved the file on different SD card. No change. So I sliced the model (coincidentaly called MINIS_bot.stl) in PrusaSlicer. No change, I could not believe my eyes.
The it hit me... as long as gcode file name contained string "MINIS", everything was printed in about 5:1 scale. As soon as I renamed the file, the print is nominal again.
So IDK, is this a feature, an easter egg or an obscure bug?

PS: No, I am not drunk.

shitcreek, 2019-10-24:
That's interesting, but could you please follow the template and provide more information such as your configuration files.

What printer are you working with and which version of Marlin are you using?

boelle, 2019-10-24:
and most important make sure you use a copy of marlin from us and not a fork somewhere else

closed 2019-10-24 by salacpavel