MarlinFirmware/Marlin #15651: TANGO (Rumba) -> canE0_AUTO_FAN_PIN and FAN_PIN

Aurawin, 2019-10-23:
#15637 # Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Aurawin, 2019-10-23:
Looking at PINS_RUMBA.h lines 143-146

#ifndef FAN_PIN
  #define FAN_PIN           7
#endif
#define FAN1_PIN            8

Line 146 (IMO) is wrong.  It should be prefaced with the conditional that it's no previously defined and should be modifiied to look like:
#ifndef FAN1_PIN
#define FAN1_PIN            8
#endif

Without these changes (and I do make them with every pull) I can't compile.  Please help remedy.

SwiftNick, 2019-10-23:
Try using undef first
```
  #undef FAN1_PIN
  #define FAN1_PIN 8
```

Aurawin, 2019-10-23:
> Try using undef first
> 
> ```
>   #undef FAN1_PIN
>   #define FAN1_PIN 8
> ```

> Executing task: platformio run <
<SNIP>
In file included from Marlin/src/gcode/control/../../inc/../pins/pins.h:150:0,
                 from Marlin/src/gcode/control/../../inc/MarlinConfig.h:32,
                 from Marlin/src/gcode/control/../gcode.h:279,
                 from Marlin/src/gcode/control/M85.cpp:23:
Marlin/src/gcode/control/../../inc/../pins/ramps/pins_RUMBA.h:147:0: warning: "FAN1_PIN" redefined
   #define FAN1_PIN            8
 ^
In file included from Marlin/src/gcode/control/../../inc/MarlinConfigPre.h:56:0,
                 from Marlin/src/gcode/control/../../inc/MarlinConfig.h:28,
                 from Marlin/src/gcode/control/../gcode.h:279,
                 from Marlin/src/gcode/control/M85.cpp:23:
Marlin/src/gcode/control/../../inc/../../Configuration_adv.h:352:0: note: this is the location of the previous definition
 #define FAN1_PIN -1
 ^
Marlin/src/gcode/control/M42.cpp: In static member function 'static void GcodeSuite::M42()':
Marlin/src/gcode/control/M42.cpp:62:9: error: duplicate case value
         case FAN1_PIN: thermalManager.fan_speed[1] = pin_status; break;
         ^
Marlin/src/gcode/control/M42.cpp:59:9: error: previously used here
         case FAN0_PIN: thermalManager.fan_speed[0] = pin_status; break;
         ^
*** [.pio/build/megaatmega2560/src/src/gcode/control/M42.cpp.o] Error 1
================================================= [FAILED] Took 3.71 seconds

SwiftNick, 2019-10-23:
Is it because  ```#define HEATER_3_PIN        8``` is using the same pin?

boelle, 2019-10-24:
if you have a working pins file just make a PR to 2.0.x so others can benefit

Aurawin, 2019-10-24:
> HEATER_3_PIN
> Is it because `#define HEATER_3_PIN 8` is using the same pin?

No, the problem is that FAN1_PIN is declared inside the there (pins_Rumba header) without the preface check to see if it was already defined - overriding my value in my Configuration_adv header.

Aurawin, 2019-10-24:
> if you have a working pins file just make a PR to 2.0.x so others can benefit

I'll go look for the other complaints about Tango previous issues.

el-quique, 2019-10-24:
Hi try:

pins_RUMBA.h
#ifndef FAN_PIN
  #define FAN_PIN           8
#endif
#define FAN1_PIN            7

Configuration_adv.h
#define E0_AUTO_FAN_PIN 7

Compile in Arduino IDE 1.8.10

boelle, 2019-10-24:
no ide 1.9.0 beta is recommended
or vscode+platform.io

boelle, 2019-10-30:
@Aurawin do you still have the isuue?

Aurawin, 2019-10-31:
I have no issues at this time.  We now have a pins_TANGO.h file to handle any previous stepper pins later.  But this is great.

PR #15682 (2019-10-28): Added BIQU TANGO V1 Support
### Description

These three changes offer support for the BIQU Tango V1 controller board.  The Tango board is based on the Rumba board with the exception of a few digital pin changes for fans and analog inputs.

### Benefits

Board output Fan 0 is now the extruder fan. 
Board output Fan 1 is now the FAN_PIN
Analog A, and Analog B digital pins are adjusted as well.
 
### Related Issues
Fixes #15651
Fixes #13567

Note: Without these changes :
1.) The installer would likely experience thermal runaway because the part cooling fan is swapped with the Extruder.  
2.) Ins [... cut]
closed 2019-10-31 by Aurawin