MarlinFirmware/Marlin #15652: [BTN_BACK isn't setting SET_INPUT_PULLUP]

ellensp, 2019-10-23:
### Description

When a display panel has a back button and it is enabled with a #define BTN_BACK {pin} in an appropriate pins.h file. The internal pull up resistor is not enabled, leaving the pin floating when not depressed. This is missing in branches 1.1.x and 2.0

### Steps to Reproduce

1. Have a display with a back button
2. attach it to a controller with software enabled pull ups  
3. add the pin the back button is on into the pin.h file for the controller (ie #define BTN_BACK {pin number}
4. Compile and upload
 
**Expected behavior:** [What you expect to happen]

Back button should work.

**Actual behavior:** [What actually happens]

Back button does not work as expected. 
Since it is floating it can be activated when the back button is not depressed. 
The physical example I had BTN_BACK was always activated, making it look like BTN_ENC was not working, when it was.

#### Solution 
In ultralcd.cpp find the following block of code  
```
  #if ENABLED(NEWPANEL)
    #if BUTTON_EXISTS(EN1)
      SET_INPUT_PULLUP(BTN_EN1);
    #endif
    #if BUTTON_EXISTS(EN2)
      SET_INPUT_PULLUP(BTN_EN2);
    #endif
    #if BUTTON_EXISTS(ENC)
      SET_INPUT_PULLUP(BTN_ENC);
    #endif
```
add the following code to enable the pull up on BTN_BACK if the BTN_BACK has been defined.
```
    #if BUTTON_EXISTS(BACK)
      SET_INPUT_PULLUP(BTN_BACK);
    #endif
```

boelle, 2019-10-24:
maybe submit a PR so it gets fixed faster?
@shitcreek

shitcreek, 2019-10-24:
I've created a PR for this issue. Please give it a test if you wouldn't mind. Thanks.

closed 2019-10-24 by shitcreek