MarlinFirmware/Marlin #15656: [BUG] LPC1769 needs a working default config on travis

LinoBarreca, 2019-10-24:
### Description

the automatic build system fails because of a wrong default config file

### Steps to Reproduce

```
In file included from Marlin/src/HAL/HAL_LPC1768/../../core/../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/../../core/serial.h:24,
                 from Marlin/src/HAL/HAL_LPC1768/DebugMonitor.cpp:26:
Marlin/src/HAL/HAL_LPC1768/../../core/../inc/SanityCheck.h:2474:6: error: #error "PSU_CONTROL requires PS_ON_PIN."
     #error "PSU_CONTROL requires PS_ON_PIN."
      ^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:25:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2474:6: error: #error "PSU_CONTROL requires PS_ON_PIN."
     #error "PSU_CONTROL requires PS_ON_PIN."
      ^~~~~
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o] Error 1
```

**Expected behavior:** regular build should succeed.

**Actual behavior:** build fails because of wrong config.

#### Additional Information

https://travis-ci.org/MarlinFirmware/Marlin/jobs/602261964

boelle, 2019-10-24:
do you know the correct config? maybe send in a PR that fixes the error?

closed 2019-10-25 by thinkyhead

thinkyhead, 2019-10-25:
77f500b8574