MarlinFirmware/Marlin #15662: [BUG] LCD manual mesh bed leveling causes reset on ESP32

vivian-ng, 2019-10-25:
When bugsfix-2.0.x is compiled for ESP32 with CR10_STOCKDISPLAY and manual mesh bed leveling enabled, using the LCD controller to start manual mesh bed leveling causes a reset of the motherboard (ESP32). The printer will home X, Y, and Z (with the message "Homing XYZ"), then, it will move the nozzle to the first mesh location. Once it reaches the first mesh location, the board resets.

["Click to begin"](http://marlinfw.org/docs/gcode/G029-mbl.html) does not show on the screen.

Stack and decoded stack trace from the ESP32:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4010f783  PS      : 0x00060930  A0      : 0x8010f7f0  A1      : 0x3ffb1e80  
A2      : 0x00000000  A3      : 0x3ffbec24  A4      : 0x00000017  A5      : 0x3ffc0ff0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e8d09  A9      : 0x3ffb1e70  
A10     : 0x0000001f  A11     : 0x3ffc0ff0  A12     : 0x00000020  A13     : 0x80000020  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4010fd70  LEND    : 0x4010fd75  LCOUNT  : 0x00000000  

Backtrace: 0x4010f783:0x3ffb1e80 0x4010f7ed:0x3ffb1ea0 0x400da7fb:0x3ffb1ec0 0x400d9ec0:0x3ffb1ef0 0x400dc7dd:0x3ffb1f20 0x400dfad9:0x3ffb1f40 0x400d3e2b:0x3ffb1f70 0x400d3fa3:0x3ffb1f90 0x400eb6c9:0x3ffb1fb0 0x4008845d:0x3ffb1fd0

PC: 0x4010f783
EXCVADDR: 0x00000000

Decoding stack results
0x4010f7ed: get_utf8_value_cb(unsigned char*, unsigned char (*)(unsigned char*), unsigned int*) at Marlin/src/lcd/fontutils.cpp line 82
0x400da7fb: utf8_strlen(char const*) at Marlin/src/lcd/fontutils.cpp line 169
0x400d9ec0: draw_edit_screen(char const*, char const*) at Marlin/src/lcd/dogm/ultralcd_DOGM.cpp line 411
0x400dc7dd: _lcd_level_bed_homing_done() at Marlin/src/lcd/menu/menu_bed_leveling.cpp line 172
0x400dfad9: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 478
0x400d3e2b: idle() at Marlin/src/Marlin.cpp line 624
0x400d3fa3: loop() at Marlin/src/Marlin.cpp line 1102
0x400eb6c9: loopTask(void*) at /home/user/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 19
0x4008845d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Based on [this](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html#loadprohibited-storeprohibited), it seems the function `get_utf8_value_cb(unsigned char*, unsigned char (*)(unsigned char*), unsigned int*) at Marlin/src/lcd/fontutils.cpp line 82` is trying to access a NULL pointer. My guess is that it is trying to access the location to display the string "Click to begin" but that somehow becomes a NULL pointer and causes a reset.

boelle, 2019-10-30:
@vivian-ng is this still an issue? and if so do you know how we could fix it? maybe we could cook up an PR

vivian-ng, 2019-10-30:
@boelle The issue remains, and I have not been able to find any clues on how to fix this. Which was why I posted this issue in the first place, to see if better programmers are able to provide some insight into solving this. Hopefully, someone more skilled and knowledgeable in Marlin and programming can offer some way to fix it.

boelle, 2019-10-30:
@shitcreek @thinkyhead does this make any  bells ring?

boelle, 2019-11-03:
@vivian-ng a lot of updates have been added, is the issue still there in 2.0.x ?

vivian-ng, 2019-11-05:
@boelle This [change](https://github.com/MarlinFirmware/Marlin/pull/15760) affects the functions which cause the reboot. I will do a pull + compile to see if it fixes the problem.

What I don't understand:
Definition of `edit_screen()`:
`void MenuEditItemBase::edit_screen(PGM_P const pstr, const char* const value/*=nullptr*/)`

But in `_lcd_level_bed_homing_done()`, it is being called with:
`MenuItem_static::draw(1, GET_TEXT(MSG_LEVEL_BED_WAITING))`

While in the function above that, it is being called in a slightly different order, such as
`MenuEditItemBase::edit_screen(GET_TEXT(MSG_MOVE_Z), ftostr43sign(v + (v < 0 ? -0.0001f : 0.0001f), '+'))`
and
`MenuEditItemBase::edit_screen(GET_TEXT(MSG_LEVEL_BED_NEXT_POINT), msg)`

vivian-ng, 2019-11-05:
@boelle @thinkyhead 
Recent changes to the LCD code seems to have solved the initial issue with LCD manual bed leveling causing reset.

However, a new problem has cropped up when trying to use LCD for leveling bed corners. After the message "Homing XYZ" on the screen, the ESP32 reboots with the following decoded stack.
```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x40084769  PS      : 0x00060336  A0      : 0x00060330  A1      : 0x3ffb0070  
A2      : 0x00000003  A3      : 0x3f401fc4  A4      : 0x0000003b  A5      : 0x3ffc0fe5  
A6      : 0x00000015  A7      : 0x3f401fbc  A8      : 0x3ffb0270  A9      : 0x3ffb0240  
A10     : 0x3f401fbc  A11     : 0x3f402c53  A12     : 0x4010f7f4  A13     : 0x3ffb0278  
A14     : 0x400d99f4  A15     : 0x3ffc160f  SAR     : 0x00000002  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x40084769:0x3ffb0070 0x0006032d:0x3ffb0150 0x400e8905:0x3ffb0170 0x400e8b41:0x3ffb01a0 0x400e8b98:0x3ffb01c0 0x400d9a19:0x3ffb01e0 0x400d9acd:0x3ffb0200 0x400d9af1:0x3ffb0240 0x400d9bf2:0x3ffb0270 0x400d8fdf:0x3ffb02b0 0x400d9e37:0x3ffb02d0 0x400db081:0x3ffb02f0 0x400dc56b:0x3ffb0310 0x400dfb39:0x3ffb0340 0x400d3e67:0x3ffb0370 0x400e3cc5:0x3ffb0390 0x400e3e2d:0x3ffb03c0 0x400e3e77:0x3ffb0400 0x400e2137:0x3ffb0440 0x400db049:0x3ffb0470 0x400dc4d0:0x3ffb04a0 0x400dc5a5:0x3ffb04d0 0x400dfb39:0x3ffb0500 0x400d3e67:0x3ffb0530 0x400e3cc5:0x3ffb0550 0x400e3e2d:0x3ffb0580 0x400e3e77:0x3ffb05c0 0x400e2137:0x3ffb0600 0x400db049:0x3ffb0630 0x400dc4d0:0x3ffb0660 0x400dc5a5:0x3ffb0690 0x400dfb39:0x3ffb06c0 0x400d3e67:0x3ffb06f0 0x400e3cc5:0x3ffb0710 0x400e3e2d:0x3ffb0740 0x400e3e77:0x3ffb0780 0x400e2137:0x3ffb07c0 0x400db049:0x3ffb07f0 0x400dc4d0:0x3ffb0820 0x400dc5a5:0x3ffb0850 0x400dfb39:0x3ffb0880 0x400d3e67:0x3ffb08b0 0x400e3cc5:0x3ffb08d0 0x400e3e2d:0x3ffb0900 0x400e3e77:0x3ffb0940 0x400e2137:0x3ffb0980 0x400db049:0x3ffb09b0 0x400dc4d0:0x3ffb09e0 0x400dc5a5:0x3ffb0a10 0x400dfb39:0x3ffb0a40 0x400d3e67:0x3ffb0a70

PC: 0x40084769
EXCVADDR: 0x00000000

Decoding stack results
0x400e8905: u8g_GetGlyph at /home/user/platformio/lib/U8glib-HAL_ID1932/src/lib/u8g_font.c line 366
0x400e8b41: u8g_font_calc_str_pixel_width at /home/user/platformio/lib/U8glib-HAL_ID1932/src/lib/u8g_font.c line 1137
0x400e8b98: u8g_GetStrPixelWidth at /home/user/platformio/lib/U8glib-HAL_ID1932/src/lib/u8g_font.c line 1183
0x400d9a19: fontgroup_cb_draw_u8g(void*, font_t const*, char const*) at Marlin/src/lcd/dogm/u8g_fontutf8.cpp line 146
0x400d9acd: fontgroup_drawwchar(font_t const*, uint32_t, void*, fontgroup_cb_draw_t, font_group_t*) at Marlin/src/lcd/dogm/u8g_fontutf8.cpp line 89
0x400d9af1: fontgroup_drawstring(font_t const*, char const*, read_byte_cb_t, void*, fontgroup_cb_draw_t, font_group_t*) at Marlin/src/lcd/dogm/u8g_fontutf8.cpp line 112
0x400d9bf2: uxg_DrawUtf8StrP(_u8g_t*, unsigned int, unsigned int, char const*, unsigned short) at Marlin/src/lcd/dogm/u8g_fontutf8.cpp line 248
0x400d8fdf: lcd_put_u8str_max_P(char const*, unsigned short) at Marlin/src/lcd/dogm/lcdprint_u8g.cpp line 51
0x400d9e37: MenuItem_static::draw(unsigned char, char const*, unsigned char, char const*) at Marlin/src/lcd/dogm/ultralcd_DOGM.cpp line 376
0x400db081: _lcd_draw_homing() at Marlin/src/lcd/menu/menu.cpp line 469
0x400dc56b: _lcd_level_bed_corners_homing() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 88
0x400dfb39: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 478
0x400d3e67: idle() at Marlin/src/Marlin.cpp line 678
0x400e3cc5: Planner::_buffer_steps(XYZEval  const&, float, unsigned char, float const&) at Marlin/src/module/planner.h line 567
0x400e3e2d: Planner::buffer_segment(float const&, float const&, float const&, float const&, float const&, unsigned char, float const&) at Marlin/src/module/planner.cpp line 2649
0x400e3e77: Planner::buffer_line(float const&, float const&, float const&, float const&, float const&, unsigned char, float) at Marlin/src/module/planner.h line 661
0x400e2137: line_to_current_position(float const&) at Marlin/src/module/planner.h line 692
0x400db049: line_to_z(float const&) at Marlin/src/lcd/menu/menu.cpp line 376
0x400dc4d0: _lcd_goto_next_corner() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 58
0x400dc5a5: _lcd_level_bed_corners_homing() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 109
0x400dfb39: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 478
0x400d3e67: idle() at Marlin/src/Marlin.cpp line 678
0x400e3cc5: Planner::_buffer_steps(XYZEval  const&, float, unsigned char, float const&) at Marlin/src/module/planner.h line 567
0x400e3e2d: Planner::buffer_segment(float const&, float const&, float const&, float const&, float const&, unsigned char, float const&) at Marlin/src/module/planner.cpp line 2649
0x400e3e77: Planner::buffer_line(float const&, float const&, float const&, float const&, float const&, unsigned char, float) at Marlin/src/module/planner.h line 661
0x400e2137: line_to_current_position(float const&) at Marlin/src/module/planner.h line 692
0x400db049: line_to_z(float const&) at Marlin/src/lcd/menu/menu.cpp line 376
0x400dc4d0: _lcd_goto_next_corner() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 58
0x400dc5a5: _lcd_level_bed_corners_homing() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 109
0x400dfb39: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 478
[... cut ...]
0x400dc4d0: _lcd_goto_next_corner() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 58
0x400dc5a5: _lcd_level_bed_corners_homing() at Marlin/src/lcd/menu/menu_bed_corners.cpp line 109
0x400dfb39: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 478
0x400d3e67: idle() at Marlin/src/Marlin.cpp line 678
```
It seems that application has written past the end of the stack (based on [this](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html#unhandled-debug-exception)).

shitcreek, 2019-11-07:
This second issue has been addressed with this commit: https://github.com/MarlinFirmware/Marlin/commit/84ce3a6bea0d12815261ef7fa40bd2c08f2a5573

closed 2019-11-07 by shitcreek