MarlinFirmware/Marlin #15668: [BUG] (short description)

tedukes, 2019-10-25:
### Description

Extruder running in reverse after upgrading from a previous version of Marlin 2.0x
### Steps to Reproduce
Building Marlin 2.0x from 10/05/19 with same settings in Marlin 2.0x from 07/08/19

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Expected to run correctly
**Actual behavior:** [What actually happens]
Extruder is running in reverse.
#### Additional Information
In the marlin release date 07/08/19 I have #define INVERT_E0_DIR true and the extruder runs properly. I set the same value in Marlin 2.0x dates 10/05/19 the same. The extruder runs in reverse. I then changed it to #define INVERT_E0_DIR false, rebuilt, and it still runs in reverse.
I tried multiple times with no success.  I have reverted back to the version from 07/08/19 and its working properly.
[Marlin07082019.zip](https://github.com/MarlinFirmware/Marlin/files/3772728/Marlin07082019.zip)
[Marlin10052019.zip](https://github.com/MarlinFirmware/Marlin/files/3772729/Marlin10052019.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-10-25:
Have you tried the most recent build?

tedukes, 2019-10-25:
No, I haven't. When I d/l the version I had intentions of working on it sooner than I did. If I have time this weekend, I can try the latest release.

tedukes, 2019-10-27:
Build today's version and everything working correctly! YEA!!

closed 2019-10-27 by tedukes