MarlinFirmware/Marlin #15670: [BUG] Fixed 3 point probing with UBL no longer works

Arti4ever, 2019-10-25:
### Description

Fixed 3 point probing with UBL no longer works. Points coordinates are not taken into account.

### Steps to Reproduce

1. Configure 3 fixed point probing with "PROBE_PT_1_X", "PROBE_PT_1_Y", etc.
2. Compile and flash
3. Do G29 J

**Expected behavior:** 
G29 J probe bed at defined fixed points.

**Actual behavior:** 
G29 J probe bed at "auto defined" points, specified points are not taken into account

shitcreek, 2019-10-25:
Mind following the template and provide your configurations?

Did you define `PROBE_PT_1_X`, etc?

Arti4ever, 2019-10-26:
Yes I defined PROBE_PT_1_X, it's part of my "step to reproduce".

Issue appear with commit e9cf46a(where custom 3 points feature was remove) and a6b0223 (where it was re-added but the wrong way).

I think this is caused by these lines in conditionals_post.h : 
> #if ALL(PROBE_PT_1_X, PROBE_PT_2_X, PROBE_PT_3_X, PROBE_PT_1_Y, PROBE_PT_2_Y, PROBE_PT_3_Y)
   #define HAS_FIXED_3POINT;
 #endif

Because test `ALL(...)` is supposed to work with defined flag not defined values.

when I bypass the condition and force `#define HAS_FIXED_3POINT;`, 3 points probing is working again.

InsanityAutomation, 2019-10-26:
I'll get a fix together tonight. Thanks for spotting this!

boelle, 2019-11-03:
@InsanityAutomation sine the PR is merged can we assume its safe to close this one?

InsanityAutomation, 2019-11-03:
yup

boelle, 2019-11-03:
and the axe is swung once again :-D

closed 2019-11-03 by boelle