MarlinFirmware/Marlin #15671: TOOL CHANGE PROCESS IN SWITCHING EXTRUDER MODE

6811620, 2019-10-25:
I use printing with two hotends according to the Dondollo scheme and SWITCHING EXTRUDER mode.
I suggest you change the tool change process (hotend) to reduce artifacts in 3D printing
When you change the hotend, the following occurs:

1) retract of the first extruder
2) lowering the bed
3) switching  hotend by servo
4) the movement of the carriage along the X axis and the Y axis to compensate offset between hotends
**5) bed lifting**
**6) the movement of the carriage along the X and Y axes to the point of continued printing in a different color**
7)  feed plastic (return from retract) by a second extruder

I am not satisfied with the sequence of events **5)** and **6)**, since after raising the table, the second hotend contacts the printed part at the point, where the first hotend stopped print. In spite of the fact that the second extruder is in a state of retract, a drop of plastic from the second hotend remains on the printed part. In addition, when the second hotend moves to the point where it will continues to print, a thin plastic thread stretches from the second hotend from contact point

My suggestion is as follows:
- the rise of the table when changing the tool should occur after the completion of the movement of the carriage to the point of continued printing with the next tool.
This should save the printed part from unnecessary artifacts.

TNX in advance
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3773157/Configuration.zip)

boelle, 2019-10-25:
is this a feature request?

6811620, 2019-10-25:
Hi, Bo Herrmannsen,

Yes, it is rather a feature request
this cannot be called a bug, because the printer is working

TNX

boelle, 2019-10-25:
oki  i have corrected the titel and added a label to reflect that

6811620, 2019-10-25:
I clearly described the problem?

boelle, 2019-10-25:
please do, but i'm just here to maintain the list of issues, i'm not a code person

AnHardt, 2019-10-25:
A perfekt job to handle in your slicer.

6811620, 2019-10-27:
@AnHardt ,
please explain what you mean
In the code that the slicer produced (Simplify3D), there are no actions to change the tool, except the T0 / T1 commands
All printer actions when changing tools are carried out by the printer's formware in the selected mode SWITCHING EXTRUDER
There are no "tool change scripts" in the slicer
Thank You

InsanityAutomation, 2019-10-27:
@6811620 
[image]

6811620, 2019-10-27:
sorry to ALL, I use **SWITCHING_NOZZLE** mode, not SWITCHING EXTRUDER.. sorry

Marlin's current release provides automatic processing of T0 and T1 commands
this automation is discussed in this topic

6811620, 2019-10-27:
for example part of code from Symplify3D

. . . 
G1 X208.627 Y103.197 F972
G92 E0
G1 E-8.0000 F900
; process PLA_0
; layer 2, Z = 0.288
**T0**                     <------------------------------------------ only needed comand to tool change
; tool H0.288 W0.500
; inner perimeter
G1 X180.250 Y109.750 F5400
G92 E0
G1 X180.250 Y130.250 E1.3006 F1215
. . .

InsanityAutomation, 2019-10-27:
Reading over the bit above, it sounds like a simple enough thing to do, and I agree offset application should be prior to or synchronous with z return. Can you confirm if you have TOOLCHANGE_NO_RETURN or TOOLCHANGE_PARK enabled? This greatly impacts the actions on tool change.

I didnt extract and read the config files to see what you have.

6811620, 2019-10-27:
@InsanityAutomation, 
if I am not mistaken, the parameters you mentioned are related to the DUAL_X_CARRIAGE mode

InsanityAutomation, 2019-10-27:
> 
> 
> @InsanityAutomation,
> if I am not mistaken, the parameters you mentioned are related to the DUAL_X_CARRIAGE mode

They are universal.

6811620, 2019-10-27:
Tell me pls in which Marlin module these parameters are determined, in the main configuration file they are not

InsanityAutomation, 2019-10-27:
@6811620 They are towards the bottom of Configuration_adv

AnHardt, 2019-10-27:
G ood part of your numbered list is no Marlin interpreting the T-code but your slicer. For that the slicer should be fixed - not Marlin.

```
G1 X208.627 Y103.197 F972  
G92 E0
G1 E-8.0000 F900  ;  1.)
; process PLA_0
; layer 2, Z = 0.288
T0  ; 2) to 5)
; tool H0.288 W0.500
; inner perimeter
G1 X180.250 Y109.750 F5400  ; 6)
G92 E0
G1 X180.250 Y130.250 E1.3006 F1215  ; 7)
```

At the moment the T0 is executed Marlin does not know about 6. Especially not if this move (if a move at all) is a extruding one or not. Only if not extruding the bed lift could be delayed.
The Z lift is entirely produced by the slicer, so it should be its decision when to drop Z again.

6811620, 2019-10-27:
@AnHardt , I understood you
the solution to my problem is to make an additional lowering of the bed **before changing the tool**, and after tool changed and arriving at the place to continue printing, **raise the bed** by the same distance
if you know te solutoin?

6811620, 2019-10-27:
the printed part will be much cleaner if you make such a permutation in the finished code

```
G1 X208.627 Y103.197 F972  
G92 E0
G1 E-8.0000 F900  ;  1.)
; process PLA_0
; layer 2, Z = 0.288
; tool H0.288 W0.500
; inner perimeter
G1 X180.250 Y109.750 F5400  ; 6)
T0  ; 2) to 5)
G92 E0
G1 X180.250 Y130.250 E1.3006 F1215  ; 7)

```
but this is a lot of manual work in the editor

6811620, 2019-10-27:
these are the artifacts I'm talking about, see below
[image]

6811620, 2019-10-28:
while I decided to use table lowering in retracts as a compromise

closed 2020-06-30 by github-actions[bot]