MarlinFirmware/Marlin #15672: No Z motion possible after homing or auto bed leveling

Louis-Cypher, 2019-10-25:
### Description

Starting with the version of bugfix 2.0 I pulled yesterday I have a very strange behaviour. I can home all axis, do auto bed leveling but when I enter a G0 or G1 move the motors are switched on but do not move. The display changes to the intended position, meaning that motion parameters seem to be completely wrong and my motor looses all steps. The strange thing is that the same motors work perfectly fine when I do homing even at decent speeds.

The behaviour described above only is present after using auto bed leveling! Until the first G29 is sent I can move Z by sending G0 or using the Move Axis menu. After sending a G29 Z doesn't move anymore at all (but I can hear that the motors are activated). There is no probing failed message or the like.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3773480/Marlin.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-10-25:
link https://github.com/MarlinFirmware/Marlin/issues/15450#issuecomment-539190835

Louis-Cypher, 2019-10-25:
You're right. Appears to be the same bug, thanks. At least I know that setting Marlin up from scratch doesn't make any sense. I'll try to use mesh leveling to see whether this is messed up as well.

boelle, 2019-10-25:
so we could in fact close this one so we dont have a dublicate?

this one will be mentioned in https://github.com/MarlinFirmware/Marlin/issues/15450#issuecomment-539190835 anyway

shitcreek, 2019-10-25:
Closing duplicate issue. Please refer to the aforementioned open issue.

closed 2019-10-25 by shitcreek