MarlinFirmware/Marlin #15673: Heating failed

3bdelli, 2019-10-25:
Hello everyone

Building a 3D Printer, using Marlin-1.1

every time I try to print, The extruder does not heating up "stays at room temperature" , while the heatbed heating up, but after 10 minutes an error appears

```
Error:Heating failed, system stopped! Heater_ID: bed
[ERROR] Error:Heating failed, system stopped! Heater_ID: bed

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```

Using MK2A heated pcb bed (200x200)

when I turn on the bed and extruder from the software "without printing", the extruder reaches the temperature that i need, while the bed keep heating up until that the same error appears "after around 10 min"

shitcreek, 2019-10-25:
Marlin 1.x is no longer supported, please try Marlin 2.0. 
To get help, please follow the template and provide your configurations.
That said, please be sure this is a bug and not a configuration error or hardware issue as we do not provide tech support here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).)

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

3bdelli, 2019-10-25:
Ok thank u
I am so sorry

closed 2019-10-25 by 3bdelli

shitcreek, 2019-10-25:
no worries. cheers