MarlinFirmware/Marlin #15674: [Reflexion] Some bugs

studiodyne, 2019-10-25:
- K value appears for each extruders ' on LCD ) in Single Nozzle mode , only one is enough
- Fan Speed/extra fan speed appears two time in single nozzle mode , but only one hotend , only one blower
- Motion sensor , is alone , but , like switch sensors , it's possible that they are multiples
- Motion sensor is under ' RUNOUT DISTANCE DEFINE'  why ?  if we want to use it as jam detection ?
**RUNOUT SYSTEM have so many bugs , but not very important for little printers , but for a MODIX 60/120, it's dramatic**
- Runout bugs:
  - RUnoutSwitch polling after block completed => if runoutdistance=25 , planner have made 24 , and the next planner block is 20 ' big outline with 0.8 nozzle ' , print dye
  - Motion sensors  ... same issue
  - If runout switch trig while a block not completed , impossible to know what distance of filament remains , so , in this case , go to pause immediatly 
  - If motions sensors have no regular polling , then you can sometimes have repetition of no change , the motion sensor must be checked every     1 pulse <Polling Motion sensor< 2 pulses , if not , impossible to have a serious detection , it's possible to have low state 1000x and no detection 

Solution ' extrusion segmentation'  in g1 command , motion sensor distance by pulse ,  and with switch , extrusion segmentation = remaining distance , no possibility to extrude more

Extrusion segmentation can make MARLIN the best secure firmware , with this we can have an very quick jam detection ' extrusion segmentation size of motion sensor pulse distance ' , and a sure runout distance of the remaining filament ' switch '

Thks

boelle, 2019-10-26:
so this is a kind of feature request?

studiodyne, 2019-10-26:
@boelle ..... May be ... but just a functionnal sketch to give inspiration ,  when i will have time for this

boelle, 2019-10-26:
its more that we have the list sorted in to either bugs or feature requests

support and discussions are meant to go on the discord server or forums

i know that there are a few open issue that fall outside that but they are sadly allready have so big roots that we cant close them before they have served their purpose

studiodyne, 2019-10-26:
For the moment , i use this thread for information , and to see if some wants to begin a starting point or some sketches of code , or may be @thinkyhead decide to apply some updates after reading this
But if no job made after a few days , i will begin my own Pull Request , and make it it progressively , by asking help to perform it to the end

boelle, 2019-10-26:
we can still call this a feature request then

if you dont mind i would change the title to [FR] RUNOUT big bugs ' in big printing' [Reflexions/Discussions]

studiodyne, 2019-10-26:
If @thinkyhead have read this thread , you can close if you want 
Or if you prefer to rename as PR , it's ok 
Thks

boelle, 2019-10-26:
there is no need to close it if it turns out to be a good idea :-D
positive thoughts from here

InsanityAutomation, 2019-10-26:
I agree this is the next logical evolution. We're at the first stage of native jam detection and what's described here is probably a stage 3. Good goal to work towards for sure.

studiodyne, 2019-11-05:
I have rewritten seriously the issue , to be understood the more seriously possible
And , by this way , a jam detection will be implemented
I have checked Stepper code , but , really too much for me , need to work six month on it to create a realistic code , with no stuttering isr , and more

Then  .... I ask for help

boelle, 2019-11-11:
@thinkyhead time have a quick look here?

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle