MarlinFirmware/Marlin #15677: [BUG] Line stair for the diagonal

Gerggg, 2019-10-26:
Since the last update 2.0, the Linear Advance with a K value greater than 0 makes 
line stair for the diagonal that make the printer vibrate unnaturally.

Do u have any idea for correction ?

Thank you.

boelle, 2019-10-26:
Try ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

please remember to follow the issue template

closed 2019-10-26 by boelle

Gerggg, 2019-10-26:
Thank you very much for your non help ...
I configure myself my marlin for over a year and I have no problem except when there are some bug.
before the PR # 10922 no problem after I can not use the Linear Advance.

I found a solution for my config but I let you look for yourself on the forum if you want.

shitcreek, 2019-10-26:
@Gerggg How did you resolve your issue? My impression was that Linear Advance doesn't work with stealthchop, and TMC2209 requires stealthchop enabled.

thinkyhead, 2019-10-26:
@Gerggg — Please be patient with our issue custodians. They do their best, and you failed to fill out the issue template according to instructions. Note that in the near future a robot will be closing issues that do not include an attachment.

As I understand it `SQUARE_WAVE_STEPPING` is very helpful for TMC steppers in-general and can resolve things like this. But we are very new to the TMC2209 and are learning which features work together and which do not. So in the near future we should be able to catch and prevent bad combinations of settings that cause these things.

Thanks for the report!

Gerggg, 2019-10-27:
@thinkyhead - I did not understand the prerequisites for opening the case. I do not speak English very well. I opened a new bug with more information.
the # 15692.

Thank you again for your work.

ManuelMcLure, 2019-10-27:
Do you know what version did not show the problem? It helps if we can narrow down the changes to a specific time period.