MarlinFirmware/Marlin #15678: [FR] Extend the M106 Command for Pre-Start Time

Powerprobot, 2019-10-26:
The slicer make one job. He adds an M106 command ahead for bridges. Then the printer print and doing his job he start the Fan and continue the way, but the air pessure isn't it fully established in this time. The air pressure need min. 1 to 4 second to stand full on object.

My idea is follow:
Extend the M106 command with a parameter "V" for "V"alue  in seconds to prestart in print process/ buffer.
In cofiguration_adv.h an addition like this:
`#define FAN_PRESTART 2000   // Add 2 Seconds to prestart the Object Fan in Printer buffer. Possible Values are 0 to 4000 (milliseconds)`

If this feature enabled, the Printer print, read the M106 command and read the following G-command. Then he placed the fan start two sceonds before the G-command start. It is might be possible, because the planner/buffer know exactly the steps to begin the G-command. So the air pressure can established full and the printhead can make the move to building the bridge.

shitcreek, 2019-10-26:
What about using G4 or M0/M1 instead? http://marlinfw.org/docs/gcode/G004.html

Powerprobot, 2019-10-26:
It is not the same. If you use G4 or M0/M1 the print ist stopped for this time, but i prefer that the print runs contiously and the fan starts a little bit before he reach the "bridge" to establish the air pressure.

thinkyhead, 2019-10-26:
At the current time this would be very tricky to accomplish, and require a re-work of the way that fans are managed in the planner. So it would be better to rely on the slicer.

profknivre, 2019-10-28:
I've created a simple post-processing script which takes care of that.

https://gist.github.com/profknivre/26d0bc812aab25117b31ccc062d558b6
https://manual.slic3r.org/advanced/post-processing

Powerprobot, 2019-10-28:
> 
> 
> I've created a simple post-processing script which takes care of that.
> 
> https://gist.github.com/profknivre/26d0bc812aab25117b31ccc062d558b6
> https://manual.slic3r.org/advanced/post-processing

Thanks for the Info. I know scripting, but i struggle sometimes...
Your Links are helpful!
i make my own "Basic" Python script:

```
#!/usr/bin/python
from __future__ import print_function
from pathlib import Path, PureWindowsPath
import sys
import re
import os
import platform

sea = re.compile(str("M106 S[1-9]+[0-9]*"))
rep = str("M106 S255\n")
repa = str("M400\n")
repb = str("G4 P1500\n")

argumentfile = sys.argv[1]
gcoderead = PureWindowsPath(argumentfile)
gcodeout = PureWindowsPath(argumentfile + "_fixed")

inname = Path(gcoderead)
outname = Path(gcodeout)

out = open(outname, 'w+')
print("Converting....")
with open(gcoderead) as f:
  for r in f:
    if re.search(sea, r) is not None:
      print (".", end = ' ')
      out.write(repa)
      out.write(r)
      out.write(repb)
    else:
      out.write(r)
```

and i add this line in Prusaslicer2:

`"C:\Program Files (x86)\Python37-32\python.exe" c:\Prusaslicer-2\Scripts\Air-pressure.py`

closed 2019-10-28 by Powerprobot

Powerprobot, 2019-10-30:
I tested my script.... it is rubbish.... it produce blobs on stops with M400. I throw it away. I need Time to write a specialized script to move the M106 some G-commands befor the bridge begin.

profknivre, 2019-10-30:
@Powerprobot my script is a bit ugly, but it does exactly what you need ;)

Powerprobot, 2019-10-30:
@profknivre : Sorry, i didn't read your script exactly before.
I read it yet and i hope i understand this script.
The script read the M106 command and place a M106 command 10 lines above with a Tag "zorq".
The Tag is need so do not double command the G-Code.
Is this correct?
it is OK but not Exactly in seconds.... "it is straight over the Thumb". Sorry for my english. :-)
I will try this script and learn more Python... to code a script to calculate the time dependent from feed rate and moved way.