MarlinFirmware/Marlin #15680: [BUG] (Unable to build SKR_Pro with current Marlin download)

ebraiman, 2019-10-26:
### Description

build fails in current version posted for SKR_Pro_v1_1

### Steps to Reproduce

1. Download and unzip Marlin bugfix -2.0.x
2. Open in PlatformIO
3. Change in configuration.h 
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1
4. Change in platforio.ini
 \default_envs = BIGTREE_SKR_PRO
5. Save and compile.

**Expected behavior:** [What you expect to happen]
Builds without issue

**Actual behavior:** [What actually happens]
Builds and fails in multiple places.

#### Additional Information
Previous downloads of Marlin with framework 5.4.3 builds without issue
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-10-26:
Use the BTT repository.

ebraiman, 2019-10-26:
Not sure I understand what you mean by using BTT repository , this originally worked nativity a couple of weeks ago.  Does BTT link there repository to Marlins?

curvi, 2019-10-26:
I think it relates to using the [BigTreeTech](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1) repo, who forked Marlin - but it is not the official Marlin repo anymore.
I would love to use the official Marlin codebase, especially since every new development and change won't be available necessarily in their repo. And according to ebraiman's youtube videos this worked not too long ago!

boelle, 2019-10-26:
> Use the BTT repository.

Personally i think this is not called for, we dont support forks so we dont ask people to go elsewhere

boelle, 2019-10-26:
> Previous downloads of Marlin with framework 5.4.3 builds without issue
>
> * Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> * Provide pictures or links to videos [...]

what is missing in the first post?

ebraiman, 2019-10-26:
The download of Marlin firmware was from www.marlinfw.org. Not sure I'm understanding your answer as I've been using for months.  A new download yesterday stopped working.  I'm thinking new SW check-in changed compiling.

boelle, 2019-10-26:
> * Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

hint hint ^^^

ebraiman, 2019-10-26:
[Marlin-bugfix-2.0.x (4).zip](https://github.com/MarlinFirmware/Marlin/files/3775194/Marlin-bugfix-2.0.x.4.zip)

This is from www.marlinfw.org

boelle, 2019-10-26:
the conig is wrong it for mega+ramps board and not SKR Pro

#define MOTHERBOARD BOARD_RAMPS_14_EFB

ebraiman, 2019-10-26:
[Marlin-bugfix-2.0.x (3).zip](https://github.com/MarlinFirmware/Marlin/files/3775199/Marlin-bugfix-2.0.x.3.zip)

[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/3775200/platformio.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3775201/Configuration.zip)
Here are the files

boelle, 2019-10-26:
i removed the first zip file as we dont need the complete marlin, just the config files

boelle, 2019-10-26:
so the changes i see in config are simple

configuration.h: 

#define SERIAL_PORT 0 
changes to
#define SERIAL_PORT -1

#define MOTHERBOARD BOARD_RAMPS_14_EFB
changes to
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1

platform.io is that also from marlin? there should only be changes to default_envs

ebraiman, 2019-10-26:
This might make it easier to follow

https://youtu.be/VsKHwYtqLz0

Just follow video with www.marlinfw.org download from today

ebraiman, 2019-10-26:
Does anyone know if bug fixes have some type of snv like number for tracking in downloads?

boelle, 2019-10-26:
was the changes i discovered correct?

ebraiman, 2019-10-26:
Yes.  except platformio.ini which was SKR_Pro specified as normal.  But the video will fill in the blanks there.

ebraiman, 2019-10-26:
I didn't want to post the video, but seeing how I've done about one a week and do not know when the change occurred in marlinfw at least you have a start and end date.

boelle, 2019-10-26:
if the platform.io comes from outside source we cant use it

but error is the same even if we use platform.io from marlin

compile error: https://pastebin.com/Y01k60KD

[Configuration&Platformio.zip](https://github.com/MarlinFirmware/Marlin/files/3775229/Configuration.Platformio.zip)

ebraiman, 2019-10-26:
Is it okay to design a build script to confirm broken builds for each board type in python?

boelle, 2019-10-26:
yes if you can, 

i think the problem is that for each of the example config there really should be an platform.io ini file
@shitcreek @thinkyhead

ebraiman, 2019-10-26:
Well depends on my time and a public server to run it on.

boelle, 2019-10-26:
there is an automatic build system testing things called travis
i think only @shitcreek and @thinkyhead can add boards and configs to that
but it would be a good way to test configs

BastR, 2019-10-27:
I have the same problem after updating the STSTM32 platform, the bug is discussed on the STSTM32 repo: [https://github.com/platformio/platform-ststm32/issues/305](https://github.com/platformio/platform-ststm32/issues/305).

I was able to compile by modifying the PlatformIO.ini file to force the use of the previous version of STSTM32:

platform = ststm32@5.6.0

curvi, 2019-10-27:
> platform = ststm32@5.6.0
This is actually already part of the current download of the Marlin 2.0 commit available on github. Still there are many errors mostly related to software-serial and compilation fails.

DroneMang, 2019-10-27:
So the current build of Marlin doesn't work with the SKR Pro?

PGrave, 2019-10-27:
Marlin 2.0 works with SKR Pro, it can be compiled (I did) using the modification above, by @BastR : 
"I was able to compile by modifying the PlatformIO.ini file to force the use of the previous version of STSTM32:
platform = ststm32@5.6.0"

Related issues: #15655, here 
[https://github.com/MarlinFirmware/Marlin/pull/15655](url)

BastR, 2019-10-27:
> > platform = ststm32@5.6.0
> 
> This is actually already part of the current download of the Marlin 2.0 commit available on github. Still there are many errors mostly related to software-serial and compilation fails.

Yes because @thinkyhead  committed yesterday : https://github.com/MarlinFirmware/Marlin/commit/e0f68cda1e0ffe4a969c34efb1d4dff893ce67b2

I tried to compile with today's build and I also have issues but with the TMCStepper library:

> In file included from Marlin/src/module/stepper/trinamic.h:29:0,
                 from Marlin/src/module/stepper/indirection.h:44,
                 from Marlin/src/module/stepper.h:47,
                 from Marlin/src/Marlin.cpp:37:
/Users/bastien/.platformio/lib/TMCStepper_ID5513/src/TMCStepper.h:26:11: fatal error: SoftwareSerial.h: No such file or directory

>   #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o] Error 1

I'm not sure what to do with SoftwareSerial.h, but by forcing the version of the TMCstepper library to the previous version too, I can compile.

Change TMCStepper@>=0.5.0,<1.0.0 by TMCStepper@0.5.1, @teemuatlut

DroneMang, 2019-10-27:
I havent tried this yet, but Bigtree says to use their version of TMCStepper: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware

DroneMang, 2019-10-27:
Here is my most up to date for an FT-6.  BL touch enabled, EEPROM enabled in configuration and flash emulation turned on in pins.  Pulling TMCStepper from Bigtree in platformio.

Errors out with Conditionals_LCD.h

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3776433/Configs.zip)

LinoBarreca, 2019-10-28:
From this morning try with https://github.com/MarlinFirmware/Marlin/pull/15655 it builds okay on Travis with default config. you **will** need to define a different timer on platformio.ini or it will hang on boot. we didn't define which one however.

boelle, 2019-11-03:
@ebraiman @DroneMang 

tried what @LinoBarreca suggested?

sjasonsmith, 2019-11-03:
@boelle, at this point it probably doesn't make sense for people to use the changes in the PR @LinoBarreca is working on, unless they are actively involved in debugging issues with the STM32 5.7 upgrade.

I am about to post a PR which should alleviate a lot of these SKR Pro / SoftwareSerial issues temporarily, until the move to 5.7+ is made. It unfortunately has to point to my own fork of a SoftwareSerial library, but that will hopefully be short-lived.

ebraiman, 2019-11-04:
This appears to be fixed with a download from Marlinfw.org with change(s) to MB for SKR Pro v1.1 and platformio.ini with serial port -1.

ebraiman, 2019-11-04:
Fixed

closed 2019-11-04 by ebraiman

yavuzongul, 2020-06-17:
I still have  similar problem. unable to build latest bugfix version of Marlin.

thinkyhead, 2020-06-17:
@yavuzongul — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).